opendatacam/opendatacam #223: Weights for yolov3-tiny and yolov3-tiny-prn aren't available on 'desktop' install

luiscosio, 2020-06-11:
Hello!

If you try to change from yolov4 to any other mode, you get the next error:

`Loading weights from yolov3-tiny.weights...Couldn't open file: yolov3-tiny.weights
`
Is there any way I can fix this?

tdurand, 2020-06-12:
Yes, they aren't included in the docker image.. I only included yolov4 for desktop : https://github.com/opendatacam/opendatacam/blob/master/docker/build/desktop/Dockerfile#L20

I need either to:

- reflect this in the documentation
- include them. which would cause a larger docker image..

I'm gonna document also how to use custom weights with the docker install of OpenDataCam... which would be a solution... you need to :

- download the weights file and put it in the same folder as the docker-compose.yml
- modify the docker-compose.yml to mount the weights at run time like this: (see the volumes part)

```yml
version: "2.3"
services:
  opendatacam:
    restart: always
    runtime: nvidia
    image: opendatacam/opendatacam:v3.0.0-beta.3-desktop
    privileged: true
    volumes:
      - './config.json:/var/local/opendatacam/config.json'
      - './yolov3-tiny.weights:/var/local/darknet/yolov3-tiny.weights'
    ports:
      - "8080:8080"
      - "8070:8070"
      - "8090:8090"
  mongo:
    image: mongo
    restart: always
    ports:
      - "27017:27017"
    volumes:
      - mongodb_data_container:/data/db
volumes:
  mongodb_data_container:
```

tdurand, 2020-06-24:
Documented now, and you have documentation on how to use other weights: https://github.com/opendatacam/opendatacam/blob/development/documentation/CONFIG.md#use-custom-neural-network-weights

closed 2020-06-24 by tdurand

luiscosio, 2020-06-24:
Excellent, thank you!