opendatacam/opendatacam #224: Data window doesnt display data

rantgithub, 2020-06-12:
Hi

This is an error I have been able to duplicate in a couple of machines

I download odc, and perform a clean install.

 I set an counter and all works fine while is counting, but when I stop the counter and click data to check, I get the following error

MJPEG-stream sent. 
known client: 53, sent = 192115, must be sent outlen = 192115
(node:18433) UnhandledPromiseRejectionWarning: MongoServerSelectionError: getaddrinfo ENOTFOUND mongo mongo:27017
    at Timeout.waitQueueMember.timer.setTimeout [as _onTimeout] (/home/odc/node_modules/mongodb/lib/core/sdam/topology.js:430:30)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
(node:18433) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 20222)

and the screen remains white.

If i press the record button, i can see the data fine, but once i stopped the screen remains white and the console shows this error

tdurand, 2020-06-12:
Yes, sorry you ran into this should have mentionned that I changed the default mongoDB url to something that works with docker

If you use OpenDataCam you need to manually change this url back to 127.0.0.1

See documentation: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#4-install-opendatacam

- Change `"MONGODB_URL"` in `opendatacam/config.json` (default is the docker service mongo URL)

```json
{
  "MONGODB_URL": "mongodb://127.0.0.1:27017"
}
```

closed 2020-06-12 by tdurand

rantgithub, 2020-06-12:
:)

I was wondering about that :)

yes, once i changed works fine :)

Tx