opendatacam/opendatacam #225: Export database for no free space

ItelBz, 2020-06-15:
Hi,
I have opendatacam installed on a Jetson Nano, it works fine. Now I have no more free space on Jetson and I would to export the database to not lost all the data.
How can I do it and then empty the database?

Best regards.

tdurand, 2020-06-15:
Hi,

The database is called "opendatacam" in the mongodb, so just google around on how to backup a mongodb database ;-)

Then if you'd like to add some documentation from your learning in the README would be also nice.

closed 2020-06-15 by tdurand

shams3049, 2020-06-15:
@ItelBz I use https://www.mongodb.com/cloud/atlas its a cloud based storage for MongoDB, There is really good documentation for using it with Node.JS applications. and there is a free tier for 500MB

ItelBz, 2020-06-17:
> @ItelBz I use https://www.mongodb.com/cloud/atlas its a cloud based storage for MongoDB, There is really good documentation for using it with Node.JS applications. and there is a free tier for 500MB

Hi, I've installed a local Mongodb on a server in the same network of the jetson and created a database called opendatacam but if i change the mongodb url in the config file it returns this error.

`(node:7327) UnhandledPromiseRejectionWarning: Error: connect ECONNREFUSED 192.168.10.203:27017
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1107:14)`

Is correct to set the url in the config file as:
`"MONGODB_URL": "mongodb://192.168.10.203:27017/"`

Thanks!