opendatacam/opendatacam #226: ODC Counter- RegisterAreas Post request example returns Server Error

Onyothi, 2020-06-18:
We would like to use API to register counter lines. However the parameters given do not match that in the request example given. When tested, Post request example given in the ODC API documentation, Counter - RegisterAreas returns server error. [https://opendatacam.github.io/opendatacam/apidoc/#api-Counter-Register_areas](url)

tdurand, 2020-06-18:
Hi, 

Could tell exactly what request are you sending so I can try to reproduce ? 

I just did a quick test and did this POST request

`POST localhost:8080/counter/areas`

body json:

```json
{
  "countingAreas": {
    "c1756828-5fb4-4174-ae0f-fdd9e90ea729": {
      "color": "yellow",
      "type": "bidirectional",
      "location": {
        "point1": {
          "x": 124,
          "y": 570
        },
        "point2": {
          "x": 247,
          "y": 564
        },
        "refResolution": {
          "w": 741,
          "h": 734
        }
      },
      "name": "test"
    }
  }
}
```

And it works

Onyothi, 2020-06-18:
Hi, 

Kindly please advise how this key in your body json was generated `"countingAreas": {
    "c1756828-5fb4-4174-ae0f-fdd9e90ea729": {`? it is different from the example given in the documentation. 

We tried the request example as is in the documentation, however that did not work

tdurand, 2020-06-18:
You can put whatever key it will work, the only requirement is to have a different for each area

I just tried the example of the doc and it works, but it was missing a comma after : `"name": "Counter line 1"`

```json
{
        "countingAreas": {
          "5287124a-4598-44e7-abaf-394018a7278b": {
            "color": "yellow",
            "location": {
              "point1": {
                "x": 221,
                "y": 588
              },
              "point2": {
                "x": 673,
                "y": 546
              },
              "refResolution": {
                "w": 1280,
                "h": 666
              }
            },
            "name": "Counter line 1",
            "type": "bidirectional"
          }
        }
      }
```

FYI I use insomnia to test API calls, very handy https://insomnia.rest/products/core/

tdurand, 2020-06-19:
closing this, feel free to reopen if you still have issue

closed 2020-06-19 by tdurand