opendatacam/opendatacam #227: Counting lines disappear when the upload/download option is used for areas

rantgithub, 2020-06-18:
Hi

When you click to either upload/download the areas and once the task is completed, the counting lines disappear from screen. 

You only see the middle boxes.

Once you press refresh, the lines show up again.

It is more a display issue (refresh) .

To reproduce, just download an area and you will see the lines disappear.

vsaw, 2022-02-13:
I don't quite understand what you mean by "Download an Area"? Could you please attach some screenshots or more detailed instructions to reproduce?

vsaw, 2022-02-13:
Actually seeing that this issue was originally reported in 2020 I assume it may have been fixed by now. I will therefore close. if you believe it is still there please reopen.

closed 2022-02-13 by vsaw