opendatacam/opendatacam #228: Using other model on top of YOLOv3 to recognize other features

MBelkhechine, 2020-06-19:
Hi all,

this is more like a question and not an issue... 

Is there any way to use any model alongside YOLOv3 to recognize other vehicle features such as make and model...

My idea is to tweak the recognition process realized by YOLOv3 of the different objects as follows:

- If YOLOv3 detects a vehicle in a frame, then with the help of OpenCV, I send the recognized vehicle (its cropped image) to another model to detect the remaining features (make and model)

- The MongoDB collection that stores the recognition data needs to be adapted to allow the persistence of these extra recognized features

Such a scenario would be preferably deployed on a server due to the additional processing steps.

Any help/guidance in the code is much appreciated... 

Thanks in advance!

tdurand, 2020-06-19:
Hi 

- You should look into darket to add this feature of running an extra model on top of YOLO on the bounding boxes .. In my opinion it is not very easy to do.. but should be possible

- Regarding hooking this to OpenDataCam, if the extra models run fast and you are able to add this extra metadata to the JSON stream you can easily persist those in the MongoDB .. this is the easy part ;-) 

Also you might be interested in looking into Deepstream v5 from nvidia which does this kind of thing.. but yes without a nice UI like OpenDataCam

Good luck !

MBelkhechine, 2020-06-19:
Thanks for your help. I will have a look at Deepstream v5 :)

tdurand, 2020-06-19:
great, closing but feel free to report anything you can share back!

closed 2020-06-19 by tdurand