opendatacam/opendatacam #229: Hi how can I use opendatacam to identify accidents in real time in a live cam feed

adnansandhila, 2020-06-19:
Hi how can I use opendatacam to identify accidents in real time in a live cam feed

Thank you
adnan

shams3049, 2020-06-19:
To my knowledge Yolo will never understand context, it does learn more about the object in context though.

This kind of problem will fall under a complete different class of ML architectures: [For example](https://machinelearningmastery.com/how-to-caption-photos-with-deep-learning/) learning to describe the images to text and then using text to identify accidents.

luiscosio, 2020-06-19:
@adnansandhila You are probably looking at something like this collision detection for OpenVino:

https://github.com/incluit/OpenVino-For-SmartCity

tdurand, 2020-06-22:
Yes, OpenDataCam isn't able to detect accidents , thanks for the pointers @shams3049 @luiscosio

closed 2020-06-22 by tdurand

adnansandhila, 2020-06-25:
Thank you everyone specially @luiscosio