opendatacam/opendatacam #231: YOLOv4 tiny released

tdurand, 2020-06-25:
I guess I'm up to re-generated v3.0.0 docker builds ;-)

https://github.com/AlexeyAB/darknet/issues/6067

Denizzje, 2020-06-25:
Would it be possible for you to make non Docker version with v4 tiny support too?

tdurand, 2020-06-25:
What do you mean ? 

I should be able to use tiny-yolov4 already now by following this doc:  https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#use-custom-neural-network-weights

Right now the default is yolov3-tiny-prn for Jetson nano, I need to redo the builds to have tinyyolov4 as a default

Denizzje, 2020-06-25:
Ah, nice. Thought you had to do an update for ODC to make it work with the new tiny model, like updating the darknet used or something. 

Seems I have something to do this weekend ;).

tdurand, 2020-06-25:
I don't think we need to update darknet code, but I will definitely need to update darknet to include the new tiny yolo .cfg file

tdurand, 2020-06-25:
> Seems I have something to do this weekend ;).

yes, I've spent half a day to generate and tag the final v3.0.0 yesterday and need to do it again ... update the docs ... bad timing...

Denizzje, 2020-06-25:
If I have a moment tomorrow I can check if it works already for non Docker ODC on Xavier NX if nessecary.

tdurand, 2020-06-25:
I just need to  double check the accuracy / speed from yolov3-tiny-prn .. it seems a little bit slower, but score 7% more on the accuracy from the benchmark .. of course welcome to give feedback

tdurand, 2020-06-26:
Just did all the builds again and made yolov4-tiny default for jetson nano build : https://github.com/opendatacam/opendatacam/releases/tag/v3.0.1

You should update your darknet to make it work if you didn't install with docker

Yolov4-tiny is very impressive in accuracy ...

closed 2020-06-26 by tdurand