opendatacam/opendatacam #239: Visual Bug

shams3049, 2020-07-14:
[image]
When the line color is black, its hard to see the directional arrow (black on black). I think the easy fix is not to have black lines.

tdurand, 2020-07-19:
Hi.. thanks, very minor but keep it on the backlog... the fix will be to just change the default from black to some other color

I'm sure you know you can add more counting lines colors in the config.json: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#customize-counter-colors

vsaw, 2021-11-12:
Fixed in development. See also https://github.com/opendatacam/opendatacam/pull/498

closed 2021-11-12 by vsaw