opendatacam/opendatacam #248: nvpmodel

zanmange, 2020-07-18:
Bonjour je souhaite utiliser opendatacam avec des powers bank ou des batteries. mais mon probleme cest que la jetson nano seteint soit au demarrage ou soit quand l'application se lance.

zanmange, 2020-07-18:
Hello I want to use opendatacam with powers bank or batteries. but my problem is that the jetson nano turns off either at startup or when the application starts.

tdurand, 2020-07-19:
You need to use it without a screen and set powermode to 5W  and it should work. See official nvidia documentation for further information: https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fpower_management_nano.html%23wwpID0E02K0HA

closed 2020-07-19 by tdurand