opendatacam/opendatacam #250: Error: Installing without docker on ubuntu 20.04 LTS

SaddamBInSyed, 2020-07-21:
Hi .

I carefully followed the steps provided [here](https://github.com/opendatacam/opendatacam/blob/development/documentation/USE_WITHOUT_DOCKER.md)

I have successfully installed darknet but when I try the node.js and MongoDB steps then I am getting the below error.

```
sudo apt-get update && apt-get install -y --no-install-recommends openssl libcurl3 mongodb-org
Hit:1 http://deb.anydesk.com all InRelease
Hit:2 http://ae.archive.ubuntu.com/ubuntu focal InRelease                      
Hit:3 http://ae.archive.ubuntu.com/ubuntu focal-updates InRelease              
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease               
Hit:6 http://ae.archive.ubuntu.com/ubuntu focal-backports InRelease            
Hit:7 https://packagecloud.io/github/git-lfs/ubuntu focal InRelease    
Ign:5 https://dl.bintray.com/etcher/debian stable InRelease
Err:8 https://dl.bintray.com/etcher/debian stable Release
  403  Forbidden [IP: 52.36.128.157 443]
Reading package lists... Done
E: The repository 'https://deb.etcher.io stable Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
amr@amr-HP-Z840-Workstation:~$ 
```

So I have installed Node.Js and Monngobd using the below commands

`sudo apt install nodejs npm
`
`sudo apt install mongodb`

I tested darknet with sample test demo script and i can able to see yolo detection video as a output.

Now, when I try to run the opendatacam then I am getting the below error,

`Process YOLO initialized
> Ready on http://localhost:8080
> Ready on http://192.168.6.110:8080
Success init db
Pheew ....! Something unexpected happened. This should be handled more gracefully. I am sorry. The culprit is:  { Error: spawn ./darknet ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:240:19)
    at onErrorNT (internal/child_process.js:415:16)
    at process._tickCallback (internal/process/next_tick.js:63:19)
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn ./darknet',
  path: './darknet',
  spawnargs:
   [ 'detector',
     'demo',
     'cfg/coco.data',
     'cfg/yolov4-416x416.cfg',
     'yolov4.weights',
     'opendatacam_videos/demo.mp4',
     '-ext_output',
     '-dont_show',
     '-dontdraw_bbox',
     '-json_port',
     8070,
     '-mjpeg_port',
     8090 ] }
Process YOLO started
(node:34323) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
`

please advise

MalcolmTomisin, 2020-08-18:
Hello, where you able to get a workaround? I am having this on Ubuntu 18.04 LTS

vsaw, 2020-08-19:
This looks like `darknet` could not be launched. I can think of two reasons

1. It is not installed
1. OpenDataCam does not find the executable in `config.PATH_TO_YOLO_DARKNET`

Without extra information is is difficult to debug this any further.

MalcolmTomisin, 2020-08-19:
> This looks like `darknet` could not be launched. I can think of two reasons
> 
> 1. It is not installed
> 2. OpenDataCam does not find the executable in `config.PATH_TO_YOLO_DARKNET`
> 
> Without extra information is is difficult to debug this any further.

I figured out darknet was not compiled properly. Unfortunately compiling darknet throws up errors 

```
compilation terminated due to -Wfatal-errors.
Makefile:168: recipe for target 'obj/convolutional_kernels.o' failed
make: *** [obj/convolutional_kernels.o] Error 1
```

vsaw, 2020-08-19:
Did you see https://github.com/opendatacam/darknet#how-to-compile-on-linux-using-make? Maybe following the compilation instructions could help...

MalcolmTomisin, 2020-08-24:
Sorry, didnt get a notification for this, apparently I needed to update the PATH and LD_LIBRARY_PATH variables as stated  [here](https://saferauto.home.blog/2019/04/06/4o-how-to-install-yolo-darknet-with-cuda-and-opencv-in-ubuntu/) . Then recompiled darknet and it did successfully. Although after running opendatacam I got a `./darknet: error while loading shared libraries: libcudart.so.10.0: cannot open shared object file: No such file or directory` . Solved that by running 

> sudo cp /usr/local/cuda-10.0/lib64/libcudart.so.10.0 /usr/local/cuda/lib/ && sudo ldconfig
sudo cp /usr/local/cuda-10.0/lib64/libcublas.so.10.0 /usr/local/cuda/lib/ && sudo ldconfig
sudo cp /usr/local/cuda-10.0/lib64/libcurand.so.10.0 /usr/local/cuda/lib/ && sudo ldconfig

P.S. If i failed to mention before. I ran it on an EC2 gpu instance with this image; Deep Learning AMI (Ubuntu 18.04) Version 32.0

tdurand, 2020-09-08:
Please follow the setup process of the README here to setup darknet: https://github.com/alexeyab/darknet

And make sure to test darknet before trying to launch OpenDataCam : https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#optional-test-darknet

closed 2020-09-08 by tdurand