opendatacam/opendatacam #252: Question on why to use MongoDB ?

ajay443, 2020-07-23:
I am trying to understand the reasons for choosing MongoDB? 
Since the target env is Jetson Nano, running a DB might impact the performance, I have seen `IOU-tracker` or `FairMOT` tracker to output the results to CSV format, here we are choosing MongoDB to persist it. My question would be, why, especially MongoDB? I believe RDBMS also works well. 

Another aspect, can someone explain how Real-Time counting works?  When is `IOU-tracker` triggered if the stream is real-time? 
I mean, say you have frames streaming into App, and you applied Yolo detections. Now I assume, after `N` frame detections, the tracker will be applied to it? When will you discard old detections, I am lost on how it works. 
Any flow chart on real-time counting?

JCrossman, 2020-08-21:
Would love some answers to this as well.

vsaw, 2020-09-01:
Running on my Jetson Nano, it does not look like Mongo has any effect on performance at all. It's mainly darknet using all of the GPU and most of the CPU resources. Please find attached the output of `tegrastats` and `top`. (Output below)

As for how the tracker works: See https://github.com/opendatacam/node-moving-things-tracker

-----------

### `tegrastats`

```
RAM 1753/3964MB (lfb 74x4MB)
SWAP 212/1982MB (cached 22MB)
CPU [26%@1479, 18%@1479, 26%@1479, 100%@1479] # <!-- CPU Load
EMC_FREQ 0%
GR3D_FREQ 99% # <-- GPU Load
PLL@35C
CPU@41C
PMIC@100C
GPU@37.5C
AO@42.5C
thermal@39.5C
POM_5V_IN 8081/8081
POM_5V_GPU 4021/4021
POM_5V_CPU 1368/1368
```

### `top`

```
  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
17393 root      20   0  9,907g 882564 271596 S 144,0 21,7   2:35.51 darknet
17362 root      20   0  920456  74804  29600 S   5,3  1,8   0:09.45 node
 4623 root      20   0       0      0      0 S   3,0  0,0 161:31.80 nvgpu_channel_p
```

tdurand, 2020-09-08:
> I am trying to understand the reasons for choosing MongoDB?
> Since the target env is Jetson Nano, running a DB might impact the performance, I have seen IOU-tracker or FairMOT tracker to output the [...]

Yes, anything would work.. We used mongodb because we were familiar with it and it worked for our use case.

> Another aspect, can someone explain how Real-Time counting works? When is IOU-tracker triggered if the stream is real-time?
> I mean, say you have frames streaming into App, and you applied Yolo [...]

- each frame we get a list of detections from YOLO
- we run a [javascript function](github.com/opendatacam/node-moving-things-tracker) (the tracker) updateWithNewFrame( YOLO Detection)  which returns the same detections with unique ids for boxes
- we run the counter algorithm to see if we count something ( a javascript function too : https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L212 )
- we persist the frame to the mongodb

The tracker keeps only a memory of 10-20 frames back and then discard everything : see code  github.com/opendatacam/node-moving-things-tracker

closed 2020-09-08 by tdurand