opendatacam/opendatacam #254: Ability to measure the speed between two counts

rjwelling2, 2020-07-29:
All,

I recently installed & configured my nVidia Jetson Nano (based on Balena) and couldn't get the CSI camera to work (which I figured out to be an issue reported here). As such I'm using a USB camera which works very well.

What would be a great feature is understand the speed that objects have while moving between 2 lines, understanding I can filter the objects (such as only cars).

Regards,
Robert 
/rw

goodengineer, 2020-07-30:
Hi, Thank you for your post.
"
 feature is understand the speed that objects have while moving
"
I need this feature as well and can collaborate on stable development.

1) Where is UML class diagram or some block-scheme / flow chart of the Opendatacam codes? If someone already did please send us a link or file to understand better the software architecture which will open more abilities/opportunities to add new features and enhance the system.
2) Do you understand how GStreamer and its plugins interact in Opendatacam when camera is connected?
3) Which camera calibration/model do you use?
4) Have you got speed estimation algorithm (pseudo-code, high-level clues) confident developed to implement further in JS/C++ or Python?

Please record video - show me how your setup is tracking cars and which data about moving cars you can extract/mine? Then on that I can analyze and we can cooperate to develop the feature.

Kind Regards,
Ing. Paul

rjwelling2, 2020-08-04:
Hi Paul, tanks for the reply. I can’t answer all of your questions as they are above my “pay grade” LOL. To answer q3, I use a Logitech C920 HD USB camera,

Let me further investigate a more suitable setup for recording a video for you to use. Cheers.
/rw

tdurand, 2020-09-08:
would need to have some API that permit to run custom logic on counting things: https://github.com/opendatacam/opendatacam/issues/258

closed 2020-09-08 by tdurand

LucasCanete, 2025-02-05:
@goodengineer Regarding your point 4. for speed estimation. I have worked on a python script that lets you measure speed in multiple areas, define the distance between lines, have the data uploaded to OpenDataCam's MongoDB and more. Take a look at it: https://github.com/LucasCanete/OpenDataCam_SpeedDetection/tree/main