opendatacam/opendatacam #257: External IP - Internal Server Error 500

effort10, 2020-08-06:
Good Day,

Has anybody been able to access opendatacam(8080) via external ip.
I keep getting the Internal Server Error 500.

All other ports work 100% (8090,8070), even 8080/status works but 8080 gives the internal server error 500.

Please could someone assist if possible.

Thank you

vsaw, 2020-08-06:
It works for me using the standard docker image.

FYI 8070 and 8090 is being forwarded to the darknet process. So all you can tell from getting results there is that darknet is running. Only connections made to port 8080 are handled by OpenDataCam itself.

Do you have any more info besides the HTTP status code? Any docker logs or similar?

effort10, 2020-08-06:
Good Day,

Thank you

I am using the standard docker image running off the Jetson Nano.
I will provide log data later this evening.

Regards

effort10, 2020-08-06:
Hi.

Please find attached the log file when i try to do a request to port 8080 outside of my network (from an external ip)
All my firewall rules have been enabled for 8080,8090,8070 as well as port forwarding.

I decided to try and run this without docker.

Regards
[logfile.txt](https://github.com/opendatacam/opendatacam/files/5035126/logfile.txt)

effort10, 2020-08-06:
Hi, I have solved this via the use of a reverse proxy.
Thank you for the assistance.

closed 2020-09-08 by tdurand