opendatacam/opendatacam #258: Alert feature: Ability to play sound when object is detected in area

tychotdeback, 2020-08-13:
Use case:

Detect potentially dangerous situation when object enters area (area defined with polygon, using development branch version of Opendatacam). For instance, say you want to prevent workers from falling in an to be avoided area at a construction site, or from coming to close to dangerous equipment. In this case it would be helpful to have a feature to play a sound to alert the admin.

tdurand, 2020-09-08:
I renamed this to add some API to hook to things.. 

right now you can do this directly in the code here: https://github.com/opendatacam/opendatacam/blob/0f921e856e7746d07a412fab8a0f7136fb004f3e/server/Opendatacam.js#L307

tdurand, 2020-09-08:
this feature would also benefit from an API hook to counter events

tdurand, 2021-05-22:
let's see if we can make this happen for next release

vsaw, 2021-05-22:
@tdurand I can give you code that fires a NodeJS event on any counter change. What would be left is the part where it calls an external webhook. Let me know if this would be of help to you.

tdurand, 2021-05-22:
I guess any input can be nice to have.. honestly I didn't think yet the potential solutions to this...

The cheap solution would be just provide a endpoint that gives all the counting events .. and people need to ping it constantly... but I'm not super familiar with technical solution to "hook" to.. websocket ?

vsaw, 2021-05-22:
Actually I just saw the eventEmitter is already part of the development branch. So all you would need is to extend server.js to connect these to Webhooks.

tdurand, 2021-05-22:
Ok thanks, from user perspective what do you think ? provide the possibility to have a simple webhook endpoint declared on the config.json ?

vsaw, 2021-05-22:
Now that I think about it this might be bigger than I first thought. 

We will need to consider

- what kind of events and data need to be transferred. 
- Rate limiting?
- Auth?

Before we can give a good estimate on the complexity and the best implementation.

We could of course make some choices but I would like to understand more how people want to use this feature to avoid wasted effort.

tdurand, 2021-05-22:
yep.. I mean, let's say if not ones gives input by the time I get on this, I will do just a super simple thing for now that can be improved later..

vsaw, 2021-05-22:
That's the point I'm trying to make. Not sure if the MVP is already too much work ;-)

munsterlander, 2021-11-11:
Somewhat unrelated / quasi-related.  I needed the ability to hook into the Jetson's GPIO pins when an object crosses a counter line.  I added a Node Python bridge so that way it just calls my Python file.  I also added a switch in the menu to turn it on and off as well as extended the Counters editor to toggle a line as an Alarm line.  So it can be done, probably not as neatly as what is being discussed here though.  In addition because I remembered seeing this thread, I went ahead and added in a node package to play configurable beeps without requiring external files.  So in short, I have an alarm that sounds on the browsers device speaker as well.  I don't think any of this code is worthy of a PR, but I do have it on the dev branch and can make a branch to contribute for review.

vsaw, 2021-11-12:
Most of the requests for Webhooks were for "live counter updates" (in some sort of dashboard)

The GPIO is new to me. I wonder if this could be implemented via the Webhooks as well? Would have the benefit to keep your fork closer to the main ODC. Or is timing an issue for you?

munsterlander, 2021-11-12:
Its possible it could be done via a webhook.  Correct me if I am wrong on this, but we are saying add an endpoint in `server.js` to be called.  Thus it would be auto-documented in API documentation and then it can be called from any external program without bogging down ODC codebase.  For the work I am doing, that "possibly" would work but I am uncertain if all the data I need would be accessible.  

In short, the current work is to call OpenCV natively in Python as I used the new polygon tool to create a quadrilateral that prompts for GPS lat/lons for known points in the camera view.  From this, we can leverage OpenCV's getPerspectiveTransform and covert any pixel coordinates into fairly accurate GPS lat/lons of the tracked item.  So what I am doing is hooking into the countItem functions to make my call, it gets the coordinates, and returns it back.  Because I have a coordinate now, we can update trackedItem with the actual coordinates and not the coordinates of the GPS receiver attached to a nano.  Additionally, if a trackedItem crosses a certain counter line, we send a PWM signal through the GPIO to attached hardware to do some downstream stuff, i.e. flash lights, activate a siren, or whatever (I can discuss this just not in a public forum)

That probably wasn't that short, but you are correct on timing may be an issue for this scenario.  Regardless of all of that, I am building on the `development` branch so once it gets released, my code "should" be able to merge and get all the benefits of the later work.

If that code could be of use to ODC, I have no issue putting it up, BUT its probably pretty hacky in terms of ODC standards, lol.

vsaw, 2021-11-12:
Actually I was thinking of doing it the other way around. Adding the option to configure a URL that gets called on every counter event and passes along details about the event (counter level, object coordinates, object type, ID etc)

That URL could of course be something like `http://localhost:<some port>` where you could have another Programm running on the same machine, receiving the counter events and doing your PWM or whatever you have in mind.

munsterlander, 2021-11-12:
Ah, ok.  That would work too - not for what I want to do - but definitely for other projects.  That makes way more sense now with your comment above:

> We will need to consider
> 
> * what kind of events and data need to be transferred.
> * Rate limiting?
> * Auth?

To be honest, I am not certain the best path forward for that type of integration because you could make a call to that url with just the current recording ID and then get all the data from the db.  Otherwise, if you are wanting to do bi-directional transactions, i.e. call and response, would we need to do a pub/sub websocket style setup.  For what I am doing, I need to alter the countedItem data for display as well within ODC.  I fear the latency in any pub/sub setup - even on the same device - would cause issues.

I do agree, this is probably edge case and not an MVP.  ODC is amazing and has accelerated my work significantly.  I can say though, the Python bridge was super easy to implement and thats also a way to offload work to external processes.

vsaw, 2022-01-25:
I decided to remove this from v4.0.0 as there is already a lot of new things in development that we need to get out.