opendatacam/opendatacam #260: Feature: Record GPS Position

vsaw, 2020-08-13:
In situation where OpenDataCam might be mobile it would be interesting to record the GPS coordinate of each counted object.

Using https://hub.docker.com/r/opensourcefoundries/gpsd and https://www.npmjs.com/package/node-gpsd I was able to quickly get GPS coordinates into POC code.

Looking at OpenDataCams architecture it should be possible to extend the `trackerEntry` and CSV export.

tdurand, 2020-09-08:
Hi, 

I saw your PR, nice ! not sure whether to include this in OpenDataCam directly as it seems a bit "niche" use case as OpenDataCam was for now though only for fixed camera.. need to think about it... it doesn't complexify much the code but you know, every feature or line of code added eventually become something to support in the future.. 

Could you tell us more about your use case ?

vsaw, 2020-09-08:
Sure, if OpenDataCam wants to be "*An open source tool to quantify the world*", we should consider that there is a lot of things to quantify that do not move, but where the camera has to move.

Here are some examples:

1. Conditions of Streets: see for example https://vialytics.de/
1. [Plastic Waste in Rivers and Harbors via Drone Footage](https://www.dfki.de/en/web/research/projects-and-publications/projects-overview/projekt/rs-of-plastic-wb1/)
1. [Early Detection of Vacant Parking Spaces Using Dashcam Videos](https://www.aaai.org/ojs/index.php/AAAI/article/view/5024/4897)

Obviously you would need custom trained models for any of those use cases but I still believe it shows a valid interest in the feature and a great improvements to open the door for OpenDataCam in those areas.

tdurand, 2020-09-08:
Thanks.. I'm not hard to convince ;-) .. Let's merge this as it does not complexify too much the code, will comment on the PR

I'm curious which is the use case you work on personally ? (it's ok if you can't share details ;-) )

vsaw, 2020-09-08:
Awesome! Yes, please share your feedback regarding the implementation with me in the PR #262. Happy to work with you in getting this upstream! 👍

tdurand, 2020-09-09:
No problem, thanks again for taking the time to share back.

closed 2020-09-09 by tdurand