opendatacam/opendatacam #268: Cannot make when CUDNN=1

watidaz, 2020-09-07:
hello, Im following the tutorial to setup ODC withought docker. However when i try to compile the darknet folder with CUDDN set to 1 i get the error below

gcc -Iinclude/ -I3rdparty/stb/include -DOPENCV `pkg-config --cflags opencv4 2> /dev/null || pkg-config --cflags opencv` -DGPU -I/usr/local/cuda/include/ -DCUDNN -DCUDNN_HALF -Wall -Wfatal-errors -Wno-unused-result -Wno-unknown-pragmas -fPIC -Ofast -DOPENCV -fopenmp -DGPU -DCUDNN -I/usr/local/cudnn/include -DCUDNN_HALF -c ./src/convolutional_layer.c -o obj/convolutional_layer.o
./src/convolutional_layer.c: In function ‘cudnn_convolutional_setup’:
./src/convolutional_layer.c:286:24: error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTEST’ undeclared (first use in this function); did you mean ‘CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3’?
     int forward_algo = CUDNN_CONVOLUTION_FWD_PREFER_FASTEST;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3
compilation terminated due to -Wfatal-errors.
Makefile:162: recipe for target 'obj/convolutional_layer.o' failed
make: *** [obj/convolutional_layer.o] Error 1

Has anyone else tun into this?
Im on Jrtpack 4.4 and running CUDA Version 10.2.89

Jeofrey

tdurand, 2020-09-08:
Hi, 

This is an issue with darknet, please ask here: github.com/alexeyab/darknet

Thanks

closed 2020-09-08 by tdurand

daliel, 2021-09-06:
faced the same problem.
downloaded the darknet from https://github.com/AlexeyAB/darknet/
changed:

```
GPU = 1
CUDNN = 1
OPENCV = 1
CUDNN_HALF = 1
arch = compute_86, code = [sm_86, compute_86]
else
LDFLAGS + = -L / usr / local / cuda-11.2 / lib64 -lcuda -lcudart -lcublas -lcurand
endif
else
CFLAGS + = -DCUDNN -I / usr / local / cuda-11.2 / include
LDFLAGS + = -L / usr / local / cuda-11.2 / lib64 -lcudnn
endif
```

run make

everything works.
downloaded the darknet from https://github.com/opendatacam/darknet
made the same changes as above.
after the make received an error:

```
[./src/convolutional_layer.c:286:24: error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTEST’ undeclared (first use in this function); did you mean ‘CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3’?
     int forward_algo = CUDNN_CONVOLUTION_FWD_PREFER_FASTEST;
                        ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3
compilation terminated due to -Wfatal-errors.
Makefile: 163: recipe for target 'obj / convolutional_layer.o' failed
make: *** [obj / convolutional_layer.o] Error 1]
```
what changes did you make? Will the odc work if I replace your darknet with AlexeyAB / darknet?