opendatacam/opendatacam #270: [Request] Possibility to stop the video if the source is a file and start recording from the beginning.

Chaaaaaaaaarlie, 2020-09-09:
I currently have some videos that I want to evaluate. But I have to draw the counting lines before I can start the evaluation. Furthermore I want to analyze the complete video and not only from the point of time when I press "Start Recording".
So the possibility to select "from the beginning" in "Start Recording" would be very convenient.

vsaw, 2020-09-09:
That is odd, what should happen is the following

1. You upload the video, the video start playing from the beginning, but it's not counting yet.
1. You switch to counter mode and define the counter lines/areas while the video plays in the background
1. You click start recording and the video playback stops and starts from the beginning

Are you sure that you are running from a video file? The behavior you describe only happens when running from "live" sources such as cameras.

Chaaaaaaaaarlie, 2020-09-09:
oh, it is already like that?
Then I'll have to recheck that.
Might be I've overlooked it.

But it's not possible to stop it, right?

vsaw, 2020-09-09:
No, a video will loop and automatically restart. However the recording stops after the first play.

closed 2020-09-09 by tdurand

zeljkomarinkovic, 2021-03-19:
Sorry for opening this issue but we have a problem when video is played from file

- drag and drop video
- place lines
- start recording
- video plays from beginning and starts counting
- video end
- loop from beginning but recording is still on

How can i stop ODP when video is done?

tdurand, 2021-03-19:
seems a bug, can you open a new issue ?