opendatacam/opendatacam #276: updateWithNewFrame is getting called twice

vladimir-kananovich, 2020-09-14:
https://github.com/opendatacam/opendatacam/blob/0f921e856e7746d07a412fab8a0f7136fb004f3e/server/Opendatacam.js#L719

Is any hidden idea behind this second call?

tdurand, 2020-09-14:
This is weird indeed ... I think the idea was to remove this  one : https://github.com/opendatacam/opendatacam/blob/0f921e856e7746d07a412fab8a0f7136fb004f3e/server/Opendatacam.js#L708

So when updateWithNewFrame fails for some reason we see the error .. otherwise it gets catch by the try catch and we don't see the full log

This was the original change: https://github.com/opendatacam/opendatacam/commit/3d930851eb9395b788a897960b5127501bbbf2ca

But then I did some revert of a feature and the updateWithNewFrame inside the try catch got back.. so it appears twice..

Very weird we didn't see this earlier I think it must not change the behaviour in some way.... 

Could you please send a PR removing the one inside the try catch? Or I will do it later...

munsterlander, 2021-11-01:
Is this still relevant?  I am getting two calls to updateWithNewFrame, but the mentioned code does not exist.

vsaw, 2021-11-04:
It does seem that this has been fixed in development in the mean time. Will close for now. Thanks for spotting this @munsterlander

closed 2021-11-04 by vsaw