opendatacam/opendatacam #280: OpenDataCam auto closed socket closed when access from internet after some minutes

dinhnn2103, 2020-09-25:
Hi there,

I am using opendatacam with jetson xavier nx and yolo v4, streaming to internet for access from external device through url (http://<myhost>:8080).

Everything seems ok but after a while (about 2-3 minutes), opendatacam stream through 8080 port killed and stuck at dark screen. Here's the log:

[image]

Access through 8090 and 8070 port (darknet) still working without problems.

Here's the commands i have used:

1. Streaming rtsp to /dev/video1 (/dev/video1 created by v4loopback)
- gst-launch-1.0 --gst-debug=3 rtspsrc location=rtsp://<myadmin>@<mypassword>@<myip>:554/Streaming/Channels/201 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! v4l2sink device=/dev/video1

2. Config to open camera from opendatacam:
 "VIDEO_INPUTS_PARAMS": {
    "usbcam": "v4l2src device=/dev/video0 ! video/x-raw ! videoconvert ! appsink",
    ...
 }

 Can somebody help me? Or how can I show log opendatacam to find out which causes this crash?

tdurand, 2020-09-29:
There is no easy way to show logs sorry.. should set debug to `true` here: https://github.com/opendatacam/opendatacam/blob/0f921e856e7746d07a412fab8a0f7136fb004f3e/server/utils/Logger.js#L5 but you need to install OpenDataCam manually for this.

If this reproducible all time ? the "hooking" mecanism to darknet can be a bit flaky sometimes.. not sure what goes wrong in your case

tdurand, 2020-11-04:
closing due to inactivity

closed 2020-11-04 by tdurand