opendatacam/opendatacam #282: CAN'T RUN DIFFERENT INSTANCES OF OPENDATACAM ON SAME HARDWARE

MalcolmTomisin, 2020-10-02:
I am trying to run different instances of ODC on an EC2 gpu instance using the  Deep Learning AMI (Ubuntu 18.04), I install without docker , run first instance of ODC and then configure to different ports , but when I try to run I get this error even after totally configuring different ports. 

> Pheew ....! Something unexpected happened. This should be handled more gracefully. I am sorry. The culprit is:  Error: listen EADDRINUSE: address already in use :::8080
    at Server.setupListenHandle [as _listen2] (net.js:1313:16)
    at listenInCluster (net.js:1361:12)
    at Server.listen (net.js:1447:7)
    at /home/ubuntu/datacam2/server.js:1062:10
    at processTicksAndRejections (internal/process/task_queues.js:97:5) {
  code: 'EADDRINUSE',
  errno: 'EADDRINUSE',
  syscall: 'listen',
  address: '::',
  port: 8080
}
Success init db

vsaw, 2020-10-02:
Looks like it did not take the new config. 

Could you share your configuration and launch commands for further debugging?

MalcolmTomisin, 2020-10-02:
I launch with `npm run start`
Here's the config straight off the console

-     Opendatacam initialized     -
- Config loaded:                  -
{
  "OPENDATACAM_VERSION": "3.0.1",
  "PATH_TO_YOLO_DARKNET": "/home/ubuntu/darknet/darknet",
  "VIDEO_INPUT": "file",
  "NEURAL_NETWORK": "yolov4",
  "VIDEO_INPUTS_PARAMS": {
    "file": "/home/ubuntu/lab-opendatacam/videos/Tuesday4August.mp4",
    "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
    "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
    "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
    "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink"
  },
  "TRACKER_SETTINGS": {
    "objectMaxAreaInPercentageOfFrame": 80,
    "confidence_threshold": 0.2,
    "iouLimit": 0.05,
    "unMatchedFrameTolerance": 5
  },
  "COUNTER_SETTINGS": {
    "minAngleWithCountingLineThreshold": 5,
    "computeTrajectoryBasedOnNbOfPastFrame": 5
  },
  "VALID_CLASSES": [
    "*"
  ],
  "DISPLAY_CLASSES": [
    {
      "class": "bicycle",
      "hexcode": "1F6B2"
    },
    {
      "class": "person",
      "hexcode": "1F6B6"
    },
    {
      "class": "truck",
      "hexcode": "1F69B"
    },
    {
      "class": "motorbike",
      "hexcode": "1F6F5"
    },
    {
      "class": "car",
      "hexcode": "1F697"
    },
    {
      "class": "bus",
      "hexcode": "1F68C"
    }
  ],
  "PATHFINDER_COLORS": [
    "#1f77b4",
    "#ff7f0e",
    "#2ca02c",
    "#d62728",
    "#9467bd",
    "#8c564b",
    "#e377c2",
    "#7f7f7f",
    "#bcbd22",
    "#17becf"
  ],
  "COUNTER_COLORS": {
    "yellow": "#FFE700",
    "turquoise": "#A3FFF4",
    "green": "#a0f17f",
    "purple": "#d070f0",
    "red": "#AB4435"
  },
  "NEURAL_NETWORK_PARAMS": {
    "yolov4": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-416x416.cfg",
      "weights": "yolov4.weights"
    },
    "yolov4-tiny": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-tiny.cfg",
      "weights": "yolov4-tiny.weights"
    }
  },
  "TRACKER_ACCURACY_DISPLAY": {
    "nbFrameBuffer": 300,
    "settings": {
      "radius": 3.1,
      "blur": 6.2,
      "step": 0.1,
      "gradient": {
        "1": "red",
        "0.4": "orange"
      },
      "canvasResolutionFactor": 0.1
    }
  },
  "MONGODB_URL": "mongodb://127.0.0.1:27017",
  "PORTS": {
    "app": 5000,
    "darknet_json_stream": 5070,
    "darknet_mjpeg_stream": 5090
  }
}

vsaw, 2020-10-09:
I see. `npm run start` executes the following line

```
PORT=8080 NODE_ENV=production node server.js
```

which sets the port to `8080`. Command line values override the config file. That explains the behaviour you see.

If you launch with

```
PORT=5000 NODE_ENV=production node server.js
```

everything should work fine.

MalcolmTomisin, 2020-10-13:
resolved! Thank you

closed 2020-10-13 by MalcolmTomisin