opendatacam/opendatacam #286: Put multiple videos of a place in a queue

pekonfuerth, 2020-10-12:
My camera setup records the video in 1h pieces. Since I have full 24h recordings, that would be 24 video files.
If I put in one file, it goes through it and finishes. When I place the next video in there, I have to reset the counter lines.
Would it be possible to make a queue for files and a checkmark if there should be new counter lines?

vsaw, 2020-10-12:
Currently OpenDataCam does not have a queue or playlist feature and I don't believe there is one in development. (You'd be welcome to start though :-))

For a workaround you can concatenate the videos before uploading it to OpenDataCam or you could try to use the [gstreamer concat plugin](https://gstreamer.freedesktop.org/documentation/coreelements/concat.html) to concatenate your files before going into Darknet.

pekonfuerth, 2020-10-12:
pfew, that's way above my skillset :l
Guess I'll just have to concatenate, as you proposed.

vsaw, 2020-10-14:
Sorry I can't do more for you. If this answers your question, would you mind closing this issue? :-)

pekonfuerth, 2020-10-15:
Aye, sorry and thanks for your reply 👍

closed 2020-10-15 by pekonfuerth