opendatacam/opendatacam #296: Using Redis instead of MongoDB

collabnix, 2020-10-21:
Opendatacam uses MongoDB as a backend database. Today Redis comes with Redis Modules like RedisJSON. The RedisJSON module is [open source](https://github.com/RedisJSON/RedisJSON/) & provides in-memory manipulation of JSON documents at high velocity and volume. With RedisJSON, you can natively store document data in a hierarchical, tree-like format to scale and query documents efficiently, significantly improving performance over storing and manipulating JSON with Lua scripts and core Redis data structures.

What do you think?

vsaw, 2020-10-21:
As far as I know it has not been tried. If you want to give it a try you would have to modify https://github.com/opendatacam/opendatacam/blob/master/server/db/DBManager.js and replace the Mongo-Connection with Redis. Rest of Opendatacam should work as is if you keep the API in place :-)

collabnix, 2020-10-22:
@vsaw Let me test it out.

tdurand, 2020-12-15:
closing this.. in summary generaly open to have different persistence layer connector.. just need to work with the same API DBManager.js if you provide a pull request with an option in the config + the documentation to switch from one or another.

closed 2020-12-15 by tdurand