opendatacam/opendatacam #297: Polygon Counter -

pekonfuerth, 2020-10-21:
I've seen you're starting to implement polygon counting.

Right now I work in a city office and most of the time we need more data than just how many people drive through a specific line.

We also have to know where people come from and what their exit is (e.g. W>E, N>S, N>W or even turnarounds like W>W). 
[image]
Here you can see on a small intersection what I mean.
Would it be possible to use the path data you already get to give out, where it starts and where it ends?
I mean it's somewhat already in the counter lines with the direction, but that's not enough :l

In the longrun we plan to use opendatacam for our permanent counting stations and use the data to control traffic lights and traffic information systems.

vsaw, 2020-10-21:
Each object tracked by opendatacam gets a unique ID. You could therefore follow the objects in the tracker and counter data to understand where it is going.

Assuming you have 4 line counters for the exits of the intersection (named left, right, top, bottom) and you get the following csv file:

frameId | timestamp | area | name | id | bearing | countingDirection | angleWithCountingLine
-- | -- | -- | -- | -- | -- | -- | --
397 | 2020-07-20T09:34:37.187Z | left | car | 1 | 191.97613244420336 | leftright_topbottom | 77.4703618136486
433 | 2020-07-20T09:34:39.346Z | right | car | 1 | 192.0947570770121 | leftright_topbottom | 77.35173718083985
450 | 2020-07-20T09:34:40.361Z | right | car | 2 | 168.95905981967627 | rightleft_bottomtop | 79.51256556182433
601 | 2020-07-20T09:34:49.659Z | top | car | 2 | 196.3895403340348 | rightleft_bottomtop | 73.05695392381718
664 | 2020-07-20T09:34:53.546Z | bottom | car | 3 | 188.61564818416412 | rightleft_bottomtop | 80.83084607368784
686 | 2020-07-20T09:34:54.888Z | bottom | car | 3 | 187.35237935989235 | leftright_topbottom | 82.09411489795959

You can see 3 cars with the IDs 1, 2 and 3 and if you follow the IDs you can see how the cars traversed the lines/exits of your intersection.

For more detailed information on their trajectories you can do a similar analysis on the tracker.json file. There, instead of counter events you can see the exact position of the object in each frame.

pekonfuerth, 2020-10-21:
hmm for that to work persistently though, the ID of a vehicle can't change and jump around as much as it is now.

At this crosswalk we wanted to count how many pedestrians use it. 
But we also simply drew another counter line on the lower dotted line.
What I noticed, however, was that many vehicles were misclassified and some of them even as pedestrians. Could this possibly have something to do with the fact that the red circled spot was recognized as a pedestrian from time to time?
Also, the ID of the vehicle coming from the north changed to another one at the traffic light. Of course this could be because the traffic light blocked the view for a short time.
[image]
The video source has roughly the same quality as the image. My RTX 2080 could process the video with ~69FPS.

vsaw, 2020-10-21:
> hmm for that to work persistently though, the ID of a vehicle can't change and jump around as much as it is now.

That is true. Have you tried improving the tracker accuracy? Here's some things you can try:

# 1 Tweak to neural network

Which neural network are you running? Can you tweak it? e.g. if you are running YOLOv4 you could try increasing the input-resolution of yolo which could help detecting smaller objects such as pedestrians. If I'm not mistaken Opendatacam sets the input resolution to 416x416px. Larger resolution e.g. 608x608px can help increase detection (See https://github.com/AlexeyAB/darknet#how-to-improve-object-detection on how to change the config)

# 2 Tweak Tracker

The node-moving-things-tracker has some [configuration options](https://github.com/opendatacam/opendatacam/blob/master/config.json#L13-L18) that could help with the tracking results.

E.g. I would check for example if the `unMatchedFrameTolerance` matches your frame rate (e.g. for how many frames is the car hidden by the sign?)

pekonfuerth, 2020-10-22:
Since I'll change my job and am not in the posession of the videos sadly I can't test that anymore, maybe you want to put this into the readme? :)

closed 2020-10-22 by pekonfuerth