opendatacam/opendatacam #300: boundix box display

belarbi2733, 2020-10-28:
Hello,

Is there any way to remove boundig boxes display but keep in it for counting ?

I don't want to see boundig box that's all

Thanks

vsaw, 2020-10-28:
Currently there is no setting to display the bounding boxes. You would have to implement it yourself. (Maybe you could also find a work around with Custom CSS overrides in the browser?)

tdurand, 2020-11-04:
yes, please have a look at the "engines" folder where we draw stuff on the canvas, you can comment out the bbox display if you like : https://github.com/opendatacam/opendatacam/blob/master/components/canvas/engines/LiveViewEngine.js 

No plan to add option for that right now

closed 2020-11-04 by tdurand

belarbi2733, 2020-11-06:
thanks for your help