opendatacam/opendatacam #303: Allow multiple client connected at the same time

vsaw, 2020-11-03:
Currently the MJPG and JSON stream only allows for one supported client at a time. Ideally we can stream the live feed to multiple connected devices at the same time.

tdurand, 2020-11-04:
Technically you need to get rid of SSE I think as it only support one connection.. and I think go with WebSocket for client - server communication ?

But also you need to take care of some complexity of having multiple client connected.. we did this one <-> one  connection at first to avoid dealing with this ;-)

vsaw, 2020-11-04:
First of all, welcome back!

I have not yet looked into SSE and do not know how we could implement mulitple active clients there. If I would tackle this I would start by moving the SSE handling into `server.js` and make `opendatacam.js` independent of the underlying stream.

One way to do this could be by using the `EventEmitter` more, as I'm currently implementing in https://github.com/opendatacam/opendatacam/pull/301

Once `Opendatacam.js` does not care about SSE or WS anymore I would try to fix the rest.

tdurand, 2020-11-04:
> First of all, welcome back!

I'll not be very active until end of the year I guess.. but trying to follow your PRs as fast as I can ;-) .. thanks for all the contributions again

> I have not yet looked into SSE and do not know how we could implement mulitple active clients there. If I would tackle this I would start by moving the SSE handling into server.js and make opendatacam.js independent of the underlying stream.

yes this would be a good first step .. hum yes.. maybe we can actually have several clients with Server Side Events.. I don't remember very well ..

> One way to do this could be by using the EventEmitter more, as I'm currently implementing in #301

I'm not very opinionated about how to implement things.. if it's easily understandable you should go for it !

tdurand, 2020-11-07:
https://github.com/opendatacam/opendatacam/pull/309  fixes this !

closed 2020-11-07 by tdurand