opendatacam/opendatacam #313: Develop can not connect to Darknet after Refactoring

vsaw, 2020-11-10:
The following lines introduced in https://github.com/opendatacam/opendatacam/pull/312 cause an error that does not allow OpenDatacam to connect to Darknet. (Unless the connection work out at the first try)

https://github.com/opendatacam/opendatacam/blob/fa5fd04c29d6c530eaa2c3d7c51ac4b4a45f0169/server/Opendatacam.js#L851-L853

tdurand, 2020-11-11:
How exactly can you repro ? maybe @mildsunrise can help ?

mildsunrise, 2020-11-11:
what's the error message?

vsaw, 2020-11-11:
The access of `err.code` fails as the parameter is called `e`.

mildsunrise, 2020-11-11:
*face-palm* sorry, I didn't notice the typo... thanks for the fix

tdurand, 2020-11-11:
always the case, even if we had 3 x 2 eyes which reviewed this PRs ;-) .. This is were typescript is handy

PR #314 (2020-11-11): 313 fix opendatacam yolo error handling
Closes #313
closed 2020-11-11 by tdurand