opendatacam/opendatacam #315: Any resolution change on Jetson Nano =

cylemmulo-1, 2020-11-11:
If I change the USB camera settings to anything but 640x360 it fails to start up after I restart docker-compose. 

 seen 64, trained: 32012 K-images (500 Kilo-batches_64) 
video file: v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=800, height=448 ! videoconvert ! appsink
[ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

I've tried 1280x720, 640x480, 800x600, 800x448 and some others. I'm using a Logitech C270 camera. If I change it back to the normal resolution it starts up fine.

Any ideas?

tdurand, 2020-11-11:
I assumed you followed https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#change-webcam-resolution and looked the video resolution supported by your webcam ? 

not sure how to help much more.. maybe you can try to debug your gstreamer pipeline outside OpenDataCam, please see: http://www.einarsundgren.se/gstreamer-basic-real-time-streaming-tutorial/

cylemmulo-1, 2020-11-11:
Yes I did double check that and was trying resolutions from there. I'll look at the debug. It's not a huge deal just would be nice to push it above the 400x400 range.  I'll give the debug a change. Maybe try out my csi camera and see if it has the same issue or not either.

vsaw, 2020-11-12:
Please make sure you read the limitations on using CSI with Docker: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#run-from-raspberry-pi-cam-jetson-nano :-)

tdurand, 2020-12-21:
closing due to inactivity

closed 2020-12-21 by tdurand