opendatacam/opendatacam #316: ANPR Integration

pigsfoot, 2020-11-19:
Hi, Sorry not really an issue but i cant find anywhere to ask questions.

Is it possible or has anyone managed to integrate ANPR capture while running Open Data Cam. We are having great success with Open Data Cam counting vehicles but woudl also like to note registration numbers so we can have two cameras and calculate journey time

shams3049, 2020-11-19:
To my understanding that goes against the principles of this project, it violates the anonymous part.

pigsfoot, 2020-11-19:
> 
> 
> To my understanding that goes against the principles of this project, it violates the anonymous part.

Not if you mask the license plate and only store hashed version.

vsaw, 2020-11-19:
There is a very old issue regarding license plates. Is this what you have in mind: https://github.com/opendatacam/opendatacam/issues/63

tdurand, 2020-11-19:
Yes, please have a look at this issue: https://github.com/opendatacam/opendatacam/issues/63

Currently nothing is beeing develop but open to contributions

closed 2020-11-19 by tdurand