opendatacam/opendatacam #318: Research to follow : Tracker "baked in" YOLO

tdurand, 2020-11-25:
https://github.com/AlexeyAB/darknet/issues/6004

Sounds very interesting as current state of the art tracking method is DeepSORT which use features from the images to perform tracking.. but is very slow cause of this..  

From what I understand there.. YOLO already plays with this features when performing object detection and the idea behind is to also use this data to perform tracking without impacting FPS / performance.

rantgithub, 2020-11-25:
I understood the same.

Yes, if this works there wont be needed the tracker because now is already built in in Yolo

haviduck, 2020-12-25:
im using a modified version of sort, its predecessor. real easy to use and no tf deps

rantgithub, 2021-01-08:
Hi

just to add the sample code mentioned on the darknet doc f you havent seen this already for detection and tracking

https://github.com/AlexeyAB/darknet/blob/master/src/yolo_console_dll.cpp