opendatacam/opendatacam #320: How to add GStreamer Plugin

jvdhurk, 2020-12-07:
Hi there! Thank you very much for this great package.

I would like to try something here. I have a monochrome machine vision USB camera (Basler dart) that I would like to use with OpenDataCam. I have already found out that this camera doesn't use v4l2src, but I have to add a specific GStreamer plugin.

Can anybody give me a hint how/where to add a GStreamer plugin? 

Thanks! 
Jan

tdurand, 2020-12-08:
If you use the  pre made docker images installation I think you can't .. My guess is you need to install [opendatacam without docker](
https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md) and install the plugin of gstreamer you need and then just change the input pipeline accordingly in the [config.json](https://github.com/opendatacam/opendatacam/blob/master/config.json#L8)

Good luck and let us know

vsaw, 2020-12-09:
@jvdhurk Can you make your camera available via RTST or similar. Then you could use the "remote_cam" pipeline. e.g. https://github.com/opendatacam/opendatacam/blob/master/config.json#L10

tdurand, 2020-12-21:
closing due to inactivity

closed 2020-12-21 by tdurand

jvdhurk, 2021-03-07:
Sorry for not getting back to you some time. The last couple of days I had time to try again and what can I say: It works!

Camera Model:
Basler daA1280-54um SMount
USB3 Boardlevel camera dart
1/3'' Aptina CMOS AR0134 **monochrom**
1280 x 960 Pixel, 54 fps
pixel size 5.5 x 5.5μm, S-Mount

How to do it:
1. Download and install: pylon 5.2.0 Camera Software Suite Linux ARM 64 Bit - Debian Installer Package
https://www.baslerweb.com/de/vertrieb-support/downloads/downloads-software/pylon-5-2-0-linux-arm-64-bit-debian/
2. Install https://github.com/joshdoe/gst-plugins-vision
3. Set path that libraries in /opt/pylon/lib64 can be found.

GStreamer Pipeline:
"baslerdart": "pylonsrc pixel-format=Mono12 fps=20 autoexposure=continuous autogain=continuous autoprofile=exposure gamma=0.7 noisereduction=1 ! videoconvert ! appsink"

tdurand, 2021-03-08:
thanks for getting back !