opendatacam/opendatacam #321: Missing GStreamer library files

palu3492, 2020-12-07:
I'm having an issue starting opendatacam when the video input is set to usbcam. The issue appears to be that GStreamer is failing to start due to missing shared object files.

Logs:
`opendatacam_1  | Loading weights from yolov4-tiny.weights...Done! Loaded 38 layers from weights-file 
opendatacam_1  | 
opendatacam_1  | (gst-plugin-scanner:151): GStreamer-WARNING **: 18:21:56.697: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstfluidsynthmidi.so': libdbus-1.so.3: cannot open shared object file: No such file or directory
opendatacam_1  | 
opendatacam_1  | (gst-plugin-scanner:151): GStreamer-WARNING **: 18:21:56.721: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstaasink.so': libslang.so.2: cannot open shared object file: No such file or directory
opendatacam_1  | 
opendatacam_1  | (gst-plugin-scanner:152): GStreamer-WARNING **: 18:21:57.771: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcacasink.so': libslang.so.2: cannot open shared object file: No such file or directory
opendatacam_1  | 
opendatacam_1  | (gst-plugin-scanner:152): GStreamer-WARNING **: 18:21:58.136: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlv2.so': liblilv-0.so.0: cannot open shared object file: No such file or directory
opendatacam_1  | 
opendatacam_1  |  seen 64, trained: 32012 K-images (500 Kilo-batches_64) 
opendatacam_1  | video file: v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink
opendatacam_1  | [ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
opendatacam_1  | [ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
opendatacam_1  | [ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created`

Are these errors indicating that the files are missing from the docker image?

tdurand, 2020-12-08:
Could you give more info about:

- Which device your are trying to run OpenDataCam ( jetson nano ? )
- Version of jetpack ? 
- Try `ls /dev/video*` to see if usbcam if detected

palu3492, 2020-12-08:
I am using a Jetson Nano with JetPack 4.3. My USB camera shows up as /dev/video0. Also, I'm using the default supplied GStreamer pipeline desc in config.json. OpenDataCam works fine with a file or an RTSP stream.

tdurand, 2020-12-08:
maybe your usb cam doesn't support this resolution.. try:

- Change webcam resolution

- First know which resolution your webcam supports, run v4l2-ctl --list-formats-ext.

- Let's say we will use 1280x720

- Change the Gstreamer pipeline accordingly: "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=1280, height=720 ! videoconvert ! appsink"

- Restart OpenDataCam

vsaw, 2020-12-09:
@palu3492 I have a similar warning when running opendatacam

```
(gst-plugin-scanner:141): GStreamer-WARNING **: 09:35:22.682: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstaasink.so': libslang.so.2: cannot open shared object file: No such file or directory
(gst-plugin-scanner:141): GStreamer-WARNING **: 09:35:23.565: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlv2.so': liblilv-0.so.0: cannot open shared object file: No such file or directory
(gst-plugin-scanner:141): GStreamer-WARNING **: 09:35:25.193: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstcacasink.so': libslang.so.2: cannot open shared object file: No such file or directory
(gst-plugin-scanner:141): GStreamer-WARNING **: 09:35:25.246: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstfluidsynthmidi.so': libdbus-1.so.3: cannot open shared object file: No such file or directory
```

Yet I can run USB-Camera without a problem. Maybe this helps excluding some of the errors. :-)

palu3492, 2020-12-15:
I'm no longer using a Nano. I'll try this on my new device in the future. Thanks.

closed 2020-12-15 by palu3492