opendatacam/opendatacam #322: YOLOv4 improvements "Scaled YOLO v4"

tdurand, 2020-12-08:
https://github.com/AlexeyAB/darknet/issues/7087

https://alexeyab84.medium.com/yolov4-the-most-accurate-real-time-neural-network-on-ms-coco-dataset-73adfd3602fe

Sseems that the summary is, yolov4 with full weight got faster + more precise .. not sure yet if tiny-yolov4 got improved too.

+ good news is fully supported by darknet , so just need to upgrade darknet to latest version + pull new weights.

tdurand, 2020-12-24:
More on this in a nice blog post: https://blog.roboflow.com/scaled-yolov4-tops-efficientdet/

vsaw, 2022-09-20:
No point in pursuing yolov4 scaled now that yolov7 is out.

vsaw, 2023-08-01:
Looking at https://github.com/opendatacam/opendatacam/issues/560#issuecomment-1660706592 I decided to re-open this issue to check if scaled yolov4 would be worth considering after all.