opendatacam/opendatacam #324: IP camera performance

geologic, 2020-12-11:
Hi

i just installed opendatacam on my 2Gb nano, i'm using jetpack 4.4 since is the one with support for 2Gb nano.
I tested it with a usb camera and it works nicely. Later i tried a 2MP IP camera, but notice a big lag between real action and video on the opendatacam live view window, sometimes more than 60 seconds of delay.
I set my IP camera to output 640x360, same resolution of the usb camera, but also tried other resolutions. IP camera and nano are on the same lan, wired with RJ-45 cable.

Anyone tried IP cameras with nano? Notice this kind of behavior?

Thanks

vsaw, 2020-12-14:
Hi,

some lag (~1s) is to be expected, even with a fast connection. However your long lag looks like there is some buffering either on the camera or on the jetson side.

Can you share your gstreamer pipeline? And have you tried to connect to the webcam from your workstation. How is the connection in that case?

geologic, 2020-12-14:
Thanks @vsaw for your reply

I just set my IP camera like this (it is a Hikvision camera):
"remote_cam": rtsp://user:pass@192.168.1.64:554/Streaming/channels/101

The camera has a admin webpage where i can live preview it on a browser, the video shows in realtime, without delay. So must be some kind of limitation/configuration on the opendatacam side.

vsaw, 2020-12-15:
Quick glance at the RTSP plugin page shows there are many options controlling the buffering and latency. Have you tried adjusting any of them? https://gstreamer.freedesktop.org/documentation/rtsp/rtspsrc.html?gi-language=c#rtspsrc

geologic, 2020-12-16:
Well, i play around with gstreamer parameters and running this on terminal i have a fluid video, without delays:

`gst-launch-1.0 rtspsrc location=rtsp://user:pass@192.168.1.64:554/Streaming/channels/101 ! latency=200 ! decodebin ! nvvidconv ! autovideosink`

But if i use it on config.json (using appsink instead of autovideosink), i still get a big delay, so i suspect it is something on the opendatacam side...

vsaw, 2020-12-16:
How big is the delay? I get around 1 sec of delay until I see the camera image in my browser. (Which is most likely a combination of darknet processing time and Opendatacam relaying the mjpg stream)

geologic, 2020-12-16:
I've playing with latency value, with latency=0 i get around 30s delay, with latency=200 around 16s. Bigger values don't improve.
After a couple of minutes opendatacam webpage restarts itself, showing "initializing opendatacam" and then goes to live preview.

At the console page i notice this:

==== HTTP Stream closed by darknet, reset UI ====
==== If you are running on a file, it is restarting  because you reached the end ====
==== If you are running on a camera, it might have crashed for some reason and we are trying to restart ====

Looks like it fills some kind of buffer and then crashes.

You say you only get 1 second delay, are you using a IP camera and nano 2Gb?

geologic, 2020-12-18:
I've made some tests to see memory usage:

- opendatacam running with USB camera and browser, memory 1360 used, 68 free, swap 100 used, 11129 free, 67% cpu
- opendatacam running with IP camera and browser, memory 1432 used, 115 free, swap 54 used, 11175 free, 167% cpu

Memory usage seems identical, cpu seems struggling to process video from IP camera.

vsaw, 2020-12-18:
Maybe different frame rates between USB and IP Camera? I don't know if you can adjust the framerate via RTSP. If not then https://gstreamer.freedesktop.org/documentation/videorate/index.html?gi-language=c could be worth a try. :-)

whiteocean016, 2020-12-21:
Hey @vsaw could you please share your solution for streaming from CSI into OpenDataCam container?
I am not familiar with gstreamer so I tried this:
launch stream on host: 
```
gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=1280, height=720, framerate=30/1' ! omxh264enc control-rate=2 bitrate=4000000 ! video/x-h264, stream-format=byte-stream ! rtph264pay mtu=1400 ! udpsink host=localhost port=5000 sync=false async=false
```
config.json:
```
"remote_hls_gstreamer": "udpsrc port=5000 ! application/x-rtp,encoding-name=H264,payload=96 ! rtph264depay ! h264parse ! queue ! avdec_h264 ! appsink"
```
and added udp port 5000 to `docker-compose.yml`, but it doesn't work (probably something wrong with gstreamer pipeline?).

vsaw, 2020-12-21:
Sending

```
gst-launch-1.0 \
        nvarguscamerasrc sensor-id=0 ! \
        "video/x-raw(memory:NVMM), width=1280, height=720, format=(string)NV12, framerate=(fraction)30/1" ! \
        jpegenc ! \
        rtpjpegpay ! \
        queue leaky=downstream max-size-buffers=1 ! \
        udpsink host=127.0.0.1 port=7001
```

Receiving

```
udpsrc port=7001 ! application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! videoconvert ! videoscale ! appsink
```

I'm not actually sure if it's enough to simple add the ports to `docker-compose`. I added the ports to the [dockerfile](https://github.com/opendatacam/opendatacam/blob/development/docker/build/nano/Dockerfile#L40) and rebuilt the image.

whiteocean016, 2020-12-21:
Sorry for hijacking the thread.

@vsaw thanks a lot for the solution, it is working now!
Although I did have to change two things:
- add `! nvvidconv` before the `! jpegenc` pipe (otherwise I got an error about jpegenc not handling caps video...)
- added port via `expose` and `ports` to `docker-compose.yml`

tdurand, 2020-12-29:
@whiteocean016 would you be open to make a Pull request to the documentation with your findings: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#run-from-ip-cam ?

geologic, 2021-01-05:
It's time to rescue my hijacked thread...
After several days of trial and error, i manage to put my ip camera and opendatacam working together. Reducing framerate to 15fps on the camera settings did the trick. Thanks @vsaw, you put me in the right direction.

closed 2021-01-05 by geologic

nishantsah08, 2022-06-04:
What did you end up using? Could you summarise the findings ?

nishantsah08, 2022-06-06:
ok i hv found the solution
json file needs to be configured with 
rtsp://admin:abcd1234@192.168.1.64/Streaming/Channels/101 ! rtph264depay ! h264parse ! nvv4l2decoder ! nvvidconv ! video/x-raw, framerate=30/1 ,width=1280,height=720 ! videoconvert ! appsink

and the fps of camera needs to be reduced. I have set it to 12. It seems that ODC is not able to fetch at higher rates. At higher rates it starts to buffer increasingly.

geologic, 2022-06-06:
@nishantsah08: glad you found the way, for me it worked just reducing frame rate at camera side to 15FPS