opendatacam/opendatacam #327: refused to start darknet or Yolo on non docker installation / jetson nano

MaximilianSteverding, 2020-12-17:
Hello everyone, I am trying to install ODC without docker on a jetson nano in order to use it with a raspberry cam. Unfortunately the installation process went smoothly when i followed the instructions on the site. Nevertheless I am running in an issue when trying to start ODC. It seems that darknet does not work somehow. Are there any hints or recommendations on how to solve this issue? 

The log looks like this:

> Ready on http://localhost:8080
> Ready on http://192.168.178.98:8080
Success init db
Pheew ....! Something unexpected happened. This should be handled more gracefully. I am sorry. The culprit is:  Error: spawn ./darknet ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:268:19)
    at onErrorNT (internal/child_process.js:470:16)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'ENOENT',
  code: 'ENOENT',
  syscall: 'spawn ./darknet',
  path: './darknet',
  spawnargs: [
    'detector',
    'demo',
    'cfg/coco.data',
    'cfg/yolov4-tiny.cfg',
    'yolov4-tiny.weights',
    'nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink',
    '-ext_output',
    '-dont_show',
    '-dontdraw_bbox',
    '-json_port',
    8070,
    '-mjpeg_port',
    8090
  ]
}
Process YOLO started
(node:6471) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
First request on console stream
Something went wrong: connect ECONNREFUSED 192.168.178.98:8070
Too much retries, YOLO took more than 3 min to start, likely an error
0
already started

Thank you very much for your help!!

vsaw, 2020-12-17:
It looks like it can not find darknet. Have you checked that "PATH_TO_YOLO_DARKNET" is set correctly?

MaximilianSteverding, 2020-12-17:
Thanks for the reply. Actually my "PATH_TO_YOLO_DARKNET" is set as suggested in the installation-guide. The current values that I changed according to the guide are:

"PATH_TO_YOLO_DARKNET" : "/home/max/darknet",
"VIDEO_INPUT": "raspberrycam",
"NEURAL_NETWORK": "yolov4-tiny",

and down below

"MONDODB_URL": "mondodb://127.0.0.1.27017",

Nevertheless I tried to dig deeper into darknet and ran the (optional) darknet test for raspberrycam within the darknet folder. It just said 
"-bash: ./darknet: No such file or directory" 
as output so I guess there might be a problem with the darknet installation right?
Do you have any suggestions how i could resolve this?

tdurand, 2020-12-17:
seems you didn't compile darknet..

Make sure you follow all the steps of the documentation: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#compile-darknet

MaximilianSteverding, 2020-12-17:
> seems you didn't compile darknet..
> 
> Make sure you follow all the steps of the documentation: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#compile-darknet

unfortunately i tried to compile darknet according to the documentation. I ran into the following error that maybe the reason for darknet beeing not compiled (printed after i executed the "make" command in the darknet folder):

chmod +x *.sh
gcc -Iinclude/ -I3rdparty/stb/include -DOPENCV `pkg-config --cflags opencv4 2> /dev/null || pkg-config --cflags opencv` -DGPU -I/usr/local/cuda/include/ -DCUDNN -Wall -Wfatal-errors -Wno-unused-result -Wno-unknown-pragmas -fPIC -Ofast -DOPENCV -DGPU -DCUDNN -I/usr/local/cudnn/include -c ./src/convolutional_layer.c -o obj/convolutional_layer.o
./src/convolutional_layer.c: In function ‘cudnn_convolutional_setup’:
./src/convolutional_layer.c:286:24: error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTES’ undeclared (first use in this function); did you mean ‘CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3’?
     int forward_algo = CUDNN_CONVOLUTION_FWD_PREFER_FASTEST;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3
compilation terminated due to -Wfatal-errors.
Makefile:162: recipe for target 'obj/convolutional_layer.o' failed
make: *** [obj/convolutional_layer.o] Error 1

If I test the nvcc version it prints out:

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Oct_23_21:14:42_PDT_2019
Cuda compilation tools, release 10.2, V10.2.89

so I expect cuda ist enabled correctly.
I am very grateful for you help!

tdurand, 2020-12-17:
Hum.. which version of jetpack are you on ? https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/FLASH_JETSON.md#how-to-find-out-my-jetpack-version

And yes this is definitely the problem

tdurand, 2020-12-17:
I think this is a problem with newer version of Jetpack which has a new version of CUDA (10.2) ... We only tested on Jetpack 4.3

Hence you have two solution:

- Instead of cloning our fork of darknet which is  , try cloning directly the latest darknet version

at [this step](https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#get-the-source-files) of the tutorial, just do this instead

```
git clone --depth 1 -b odc https://github.com/AlexeyAB/darknet
```

It might work with the new CUDA stuff, our darknet version is from late june: https://github.com/opendatacam/darknet/tree/odc  , but we have [very little difference with the main darknet](https://github.com/opendatacam/darknet/pull/7/files) , mostly you will just have more noise in the console

OR the way that we are sure work would be to:

- flash your jetson to jetpack 4.3 : https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/FLASH_JETSON.md#%EF%B8%8Fflash-jetson-board

Early next year we are gonna release a new version of OpenDataCam which will have the latest darknet code.. and will be able to use latest Jetpack version

MaximilianSteverding, 2020-12-19:
@tdurand thank you for your suggestions. I managed to get darknet running so the jetpack version is definitely the problem because I still get the same errors and ODV is not starting. 

I am currently on jetpack 4.4.1 with the jetson nano 2GB version. When I am trying to flash jetpack version 4.3 onto the jetson nano it won`t boot but prompts me a screen to choose wether to continue in save boot mode or not. Unfortunately I cannot interact with the menu because the keyboard is not recognized at that stage somehow. As long as nobody has a suggestion where to find a working jetpack 4.3 version for jetson nano 2GB I think it will rather be a topic for an nvidia forum or i will have to wait until the ODC version for jetpack 4.4.1 is released. Thank you all for your help!

MaximilianSteverding, 2020-12-19:
Just a final update on the topic:

The Jetson Nano 2GB does not support Jetpack 4.3 ([https://forums.developer.nvidia.com/t/jetpack-4-3-on-jetsonnano-2g/159624](url)). Nevertheless the above mentioned problem only appears when trying to install ODC without docker. ODC does work on the newer Jetpack Version 4.4.1 and on the Jetson Nano 2GB with a docker install. Nevertheless this means that one cannot use the raspberry cam at this point for ODC. I have not tested USB Cam yet but ODC does work on video files.

tdurand, 2020-12-20:
@MaximilianSteverding yes this is what I suspected.. I will let this issue open as we need to work on updating OpenDataCam for latest jetpack version.

But:

> I managed to get darknet running so the jetpack version is definitely the problem because I still get the same errors and ODV is not starting.

So if you managed to get darknet starting (I guess the alexeyAB latest version ?)..  ODC should start if you point the PATH_TO_DARKNET to this darknet version working.. you have the same error ?

In timings, I aim to dedicated some time late january to release a new version of OpenDataCam

MaximilianSteverding, 2020-12-20:
@tdurand I managed to install the alexeyAB latest version and got darknek running but when pointing ODC to the darknet version with the PATH_TO_DARKNET variable it resulted in the same issue as with the darknet version that you deploy. I will try to recreate the error (as i have flashed the installation when i tried installing the older jetpack version) and see wether i can report any changes in the log. 
Thanks for the updated regarding the schedule. Maybe I can contribute with the error logs or get darknet to communicate with ODC.

tdurand, 2020-12-20:
Yes, I think if darknet can work, OpenDataCam should work unless darknet changed some other stuff (the way we integrate with it) since then, but I don't believe so.

vsaw, 2021-12-08:
Will close this as JetPack 4.6 is now the most current. And I just tested that the 2 GB Jetson nano works with JP 4.6 so I consider this obsolete now.

closed 2021-12-08 by vsaw