opendatacam/opendatacam #334: Generate API on build

vsaw, 2020-12-19:
As far as I can tell, the whole API Documentation can be generated with `npm run generateapidoc`. Therefore it should be considered build artifacts and not be comitted to git.

Ideally, `npm run build` and `npm run dev` should build the apidoc automatically when executed.

vsaw, 2020-12-21:
Closed in https://github.com/opendatacam/opendatacam/pull/336

tdurand, 2020-12-21:
I'll reopen, as we will need to double check if this works when we merge development into master at some point .. Will probably need to push two commits, the first one will create the gh-pages branch, and then we will need to select the gh-pages branch in the settings as the branch for Github Pages . (currently it is set to master)

Let me know if I missed something

vsaw, 2020-12-21:
You can already prepare the repo to serve from `gh-pages`. Just create a copy of the master branch and change the settings for the GitHub Pages. This way, once we merge development into master, the action can just run.

You are right that we need to verify that it runs correctly. In my opinion though this feels more like a separate task as this task was to remove the `apidoc` folder from git (which we did 🎉)

tdurand, 2020-12-22:
true, will do that and put a note in the development notes

PR #336 (2020-12-21): Generate apidoc on build
Closes #334
closed 2020-12-22 by tdurand