opendatacam/opendatacam #337: Code-Quality and  -Architecture Discussion

vsaw, 2020-12-19:
I don't know where else to put it, but I would like to start a conversation regarding Opendatacam's code-quality and -architecture.

Frankly I believe ODC would benefit greatly from improvements in code-quality and -architecture. I've started to put ODC on that path by adding Jasmine tests and eslint rules. While this will hopefully improve the code-quality over time, the architecture still needs guidelines.

I personally see the following to fix

- Darknet "leaks": To allow us to plug in different Networks we should aim to contain all Darknet related code in [YoloDarknet.js](https://github.com/opendatacam/opendatacam/blob/development/server/processes/YoloDarknet.js).
- Counter Logic: We've already seen massive work being done on the counter logic by adding the polygon counter. Currently this bloats [Opendatacam.js](https://github.com/opendatacam/opendatacam/blob/development/server/Opendatacam.js). We could try to carve this out before we add "speed measurements" and other features
- Make [Opendatacam.js](https://github.com/opendatacam/opendatacam/blob/development/server/Opendatacam.js) agnostic of SSE and other underlying networks. Ideally there should be a spearation of concern between the Network connectivity and Opendatacams logic.
- Database Abstraction. At the moment all code accesses the MongoDB Abstraction directly, which makes it difficult to change to different persistence as currently being developed in the [evo-addnedbconnector](https://github.com/opendatacam/opendatacam/tree/evo-addnedbconnector) branch
- Direct include of [config.json](https://github.com/opendatacam/opendatacam/blob/development/config.json) makes it difficult during development to try out different configs quickly. Additionally it makes it impossible to change configurations at runtime.

While I'm sure we could extend the list I am wondering how the other's (e.g. @tdurand or @mildsunrise) think about this and if we should coordinate our efforts.

tdurand, 2020-12-20:
@vsaw ! Many thanks for this !! 

All the points are 👌 ! totally agree !

Actually there is some exciting news around the OpenDataCam project we didn't share publicly yet but we are working on release a version for Mobile, running on Android (running the YOLO inference too there).. And part of the work on this we need to make OpenDataCam more "darknet" agnostic and more modular to be able to have a build for Android, a build for Darknet , etc etc..

Please ping me on my email / twitter ( on my github profile you'll find it ) if you are around this week before christimas to schedule a quick call to synchronize efforts so we don't refactor the same things and run into merge problems ! Will post the time on github so anyone interested can join too.

tdurand, 2020-12-21:
Tomorrow 12 A.M (Europe time) this is happening, if anyone interested please ping me on my email and I'll add you in the meeting.

vsaw, 2020-12-22:
We discussed during the call to proceed as follows:

- There will be no "big rewrite" instead we will incrementally refactor the parts individually
- @vsaw create separate issues for each of the points mentioned above to keep track of the refactoring progress
  - [Refactor: Darknet "leaks"](https://github.com/opendatacam/opendatacam/issues/342)
  - [Refactor: Counter Logic](https://github.com/opendatacam/opendatacam/issues/343)
  - [Refactor: Network agnostic Opendatacam.js](https://github.com/opendatacam/opendatacam/issues/344)
  - [Refactor: Better Database abstraction](https://github.com/opendatacam/opendatacam/issues/345)
  - [Refactor: Better Configuration Handling](https://github.com/opendatacam/opendatacam/issues/346)
- @vsaw will start refactoring the DB system https://github.com/opendatacam/opendatacam/issues/345
- @tdurand will continue developing the mobile version of Opendatacam on the mobile branch.
- @tdurand will rebase and merge his mobile branch with development after mobile development finished

Specifics regarding the work to be done can be discussed in the respective issues. The "general discussion" is considered finished for now 👍

closed 2020-12-22 by vsaw