opendatacam/opendatacam #349: Reconnecting to External Database upon losing connection

uvaaland, 2020-12-26:
Hi!

I have successfully run OpenDataCam using an external MongoDB server by altering the URL in the config file. However, when I tested to see what would happen if the connection to the database client was lost, I noticed that OpenDataCam would not try to reconnect to the client and I had to restart the service to get it running properly again.

This means that any data collected after the connection to the database was lost would also be lost, which is undesirable. 

Am I understanding it correctly that there is no functionality in OpenDataCam at the moment to try and re-establish the connection to the database client if it is lost? And do you see an easy way to add this functionality if I desire?

Thanks! 😄

vsaw, 2020-12-27:
You are right. Opendatacam was built with a local Database in mind, therefore connection issues are not expected and not handled.

However I'm currently working on refactoring the database subsystem in https://github.com/opendatacam/opendatacam/pull/347. This would allow us to easily improve the database connection and implement auto-reconnects. If you want to give it a go feel free to use the code in https://github.com/opendatacam/opendatacam/pull/347 and add to it. I believe you could implement the auto-reconnect entirely in the [`MongoDbManager.js`](https://github.com/opendatacam/opendatacam/pull/347/files#diff-2ce65e583991c79a19bf86254cf9263428b0c900936e77d3fd97c3d42491d260) file. Otherwise give me some time an eventually I'll get around to to it :-)

uvaaland, 2021-01-06:
Hi, Valentin!

Thanks for explaining. The feature isn't essential to what I am working on, but I wanted to make sure that I understood the current implementation correctly.

I see here that you already have a commit in the works that addresses the issue, great! 😄 👍

vsaw, 2021-01-06:
@uvaaland Yes I'm currently working on a fix. Once ready I will publish a "development" docker image for you to try if you like :-)

vsaw, 2021-01-08:
@uvaaland I've pushed the fix to https://hub.docker.com/r/vsawdoteu/opendatacam/tags?page=1&ordering=last_updated feel free to give it a spin. You can download it once [this action](https://github.com/vsaw/opendatacam/actions/runs/472830282) has finished.

Please note that the config has slightly changed. See https://github.com/vsaw/opendatacam/blob/349-mongo-db-reconnect/documentation/CONFIG.md#database on how to configure it.

vsaw, 2021-01-12:
Closed via https://github.com/opendatacam/opendatacam/pull/361

PR #361 (2021-01-12): Reconnect to MongoDB on connection loss
Closes #349
closed 2021-01-12 by vsaw