opendatacam/opendatacam #350: OpenDataCam for Bike Project

willyjtc, 2020-12-28:
Hello,

I am working on a project to count the number of bikes on a road and I am using Jetson Nano. 

I need that the app works a number of hours a day and triggers an alert if there are more than x bikes on the road. I'm having a lot of trouble starting Jetson Nano with a Rasberry Pi camera, but I hope to achieve it (I don't know if I should use Docker or not). 

I have read that the performance of the Jetson Nano is not efficient. What do you recommend? Jetson change?

Thanks for your great work

tdurand, 2020-12-29:
Hi, 

Unfortunately right now the Docker install of OpenDataCam does not support Raspberry Pi Cam , please follow https://github.com/opendatacam/opendatacam/issues/178 for updates .. If you want to use Raspberry Pi Cam you should install OpenDataCam without docker

Also we have issues with Jetson nano 2GB , which doesn't seems to be compatible yet: https://github.com/opendatacam/opendatacam/issues/327

Jetson nano will run perfectly the tinyYOLOv4 model, which should be good enough to detect bikes, but obviously the better model you can run the best accuracy you get... so if you can afford a Xavier for your project it will be better.

I would say first validate your project with Jetson Nano and them upgrade Jetson if needed.

closed 2021-01-08 by tdurand

willytrueba, 2021-04-21:
Hello again,

Thank you first for answering and your time. I'm still working on the bike counting project, but I run into several problems. At the moment I use a Jetson Nano with USBCam, and after installing it, everything works ok a couple of times. If I turn off and on a couple of times I have the following error and I don't know how to solve it.

```
Start recording
already started
First request on console stream
Something went wrong: connect ECONNREFUSED 172.18.0.3:8070
Too many retries, YOLO took more than 3 min to start, likely an error
0
Something went wrong: connect ECONNREFUSED 172.18.0.3:8070
Too many retries, YOLO took more than 3 min to start, likely an error
0
New client, close previous stream
New client, close previous stream
already started
Something went wrong: connect ECONNREFUSED 172.18.0.3:8070
Too many retries, YOLO took more than 3 min to start, likely an error
0
```
Also the Ubuntu terminal does not work either ...

My configuration is this:

I am working so that a daemon every x minutes uploads the data to a server and clears the logs, so that ODC is not loaded with much information.

You recommend that I change Jetson, or use another tool. And is ODC going to have a version soon?

Greetings and thank you