opendatacam/opendatacam #355: Has anyone (like me currently) worked or is working on cleaning up the results

hacktisch, 2021-01-03:
I cooperate with somebody who provides me with JSON exports he makes using OpenDataCam, of traffic.
I In this project, we need data that is cleaner than currently exported from OpenDataCam, which is why I am correcting these data errors:

1. ) Incorrectly splitting a single object into multiple objects (one car gets assigned more than one ID over the course of its travel), for example when it drives past behind a traffic light. I now extrapolate the tracks with some linear regression trickery and then map the objects with most likely matches.
2. ) Incorrectly merging objects as if they were a single object, for example when two cars drive alongside each other. I am now un-merging these cars and adding them as separate objects each with their own track. I do this by checking sudden accelerations, too large jumps etc. And see if I can form a new line from the resulting coordinate points, to conclude whether that is indeed another object.

Now I'm half way with writing a cleanup script and I think it will work fine when I'm done; but it just now occurred to me to ask the community whether someone already built a thing that does exactly what I'm doing (and which probably does it better than I do).

tdurand, 2021-01-03:
Super interesting that your are working on improving the data based on the specific use case of tracking cars and traffic.. If you can please share it with the community afterwise and I'll make sure to reference it on the documentation.

Also in the future we hope to be able to provide improvements to the tracker: https://github.com/opendatacam/node-moving-things-tracker , so you have less stuff to clean up 😉

vsaw, 2021-01-03:
@hacktisch Have you looked into the https://github.com/opendatacam/node-moving-things-tracker maybe instead of cleaning up the results you can fix the tracker so they come out correctly?

tdurand, 2021-01-03:
From what I understood he is doing some specific tweaks that applied to cars.. You can make some assumption that no object are coming up from the sides,  no jump in trajectory,  etc etc.. and unfortunately you can't do this for everything as this might hurt the performance of the tracker for other use cases.. And also as the clean up is done after the recording, you have a access to information you don't have when you perform the tracking real time. (you can analysis the whole trajectory)

That said, there might be ideas to include in the tracker directly 😉

rantgithub, 2021-01-03:
I have tested the current tracker extensively.
There are some limitations coming from yolo and one of those is the identification of objects when they are very close. The tracker will be limited on what it can do depending what is the detection coming from yolo
Cleaning the data is a way to improve results but the tracker makes the difference.

hacktisch, 2021-01-03:
Hey all,
Thanks for all your quick replies. I hadn't looked at node-moving-things, I haven't even run OpenDataCam yet myself, but I might try again soon (I was struggling to set it up on my machine and then decided to just use the JSON exports my partner made).

@tdurand, the cleanup that I do is indeed for cars, and may not be suitable for 100% of the situations, but I do think it would be helpful in most cases. In most cases in the physical world, objects do not suddenly disappear (and another object magically appears at the same spot 1 frame later). And also most times objects don't teleport around the scene.

If you are unsure whether you want to add these kinds of fixing to node-moving-things because it may not apply to all use cases, you could add it an optional parameters. After taking a quick look at the repository, I think I understand correctly that the main classifier for determining whether an object in frame 1 is the same object as in frame 2, is the amount of overlapping surface of the rectangle you receive from object recognition from YOLO. I suppose that another way of drastically improving the data quality is to compare the contents of the image inside the rectangle between frame 1 and 2 (it looks as this isn't done currently?) if the images are very dissimilar, they probably were different objects.
And matching objects based on the overlapping surface can be problematic when for instance you have a thin object that is moving sideways, then it doesn't have any overlapping surface between frames.

What I was writing myself now is very scripty and hacky, as we only needed to have clean data first to be able to test a hypothesis of ours. If it proves to work, I will probably rewrite it properly, and then share it with you. But I can write pseudocode of what I do.
However I'm not sure how useful it will be for you. I'm sure that there's plenty of academic papers where better methods are proposed than anything I could invent now as I'm probably trying to reinvent the wheel for most things.

tdurand, 2021-01-03:
thanks for the input, and yes I think just validate something first with your project and then we can see if it makes sense to upstream it.. or only put it as a reference in the documentation

Some comments:

> compare the contents of the image inside the rectangle between frame 1 and 2 (it looks as this isn't done currently?) if the images are very dissimilar, they probably were different objects.

Yes this is how you improve things but it is very slow to do.. The "state of the art" is currently DeepSORT but we can't run it real time, some alternative approach we are watching are: tracker "baked in" yolo, https://github.com/opendatacam/opendatacam/issues/318 , fairMOT https://github.com/opendatacam/opendatacam/issues/195

> I'm sure that there's plenty of academic papers where better methods are proposed than anything I could invent now as I'm probably trying to reinvent the wheel for most things.

Yes , if you are interested look into the https://motchallenge.net/

I'm pretty busy right now but by end of january I hope to have time to dedicate to improve the tracker.

rantgithub, 2021-01-03:
ODC has the "interface" on top what YOLO does,
As research, just load the video you are trying to perform the analysis in yolo alone and you can see how the detection plays.
After that any tracker can be applied, but again there are some limitation of what yolo can do. alternatives neural networks or parallel can be used to help with the problem of objects very close, but that is something needs to be done outside of odc

closed 2021-05-22 by tdurand