opendatacam/opendatacam #358: CI/CD: Use pre-build Opendatacam Darknet Images

vsaw, 2021-01-03:
Currently the CI/CD pipeline builds https://github.com/opendatacam/darknet from source every time. As Darknet is not going through many changes we could create a Darknet Docker Image for Opendatacam and then use this image directly.

It should cut off approx 5 minutes of execution time from the Workflow.

# See also

- https://github.com/opendatacam/opendatacam/pull/354

vsaw, 2021-01-14:
@tdurand If you have some spare minutes, I would appreciate you preparing the darknet repository so I can use a "pre-built" darknet in our docker images. See https://github.com/opendatacam/darknet/pull/8

tdurand, 2021-01-14:
@vsaw I gave you full access to the repo, let me know if you can do it yourself or if you still need me :-)

vsaw, 2021-01-14:
@tdurand Thank you. If you can set up the `DOCKERHUB_USERNAME` and `DOCKERHUB_TOKEN` secrets I can take care of the rest. (See https://docs.docker.com/docker-hub/access-tokens/ how to create tokens).

tdurand, 2021-01-14:
Ok, sorry was not around.. adding it now

tdurand, 2021-01-14:
done 👍

vsaw, 2021-01-15:
Thanks. I'm currently testing the newly created development branch here: https://github.com/opendatacam/darknet/actions/runs/487387229

If this works I will merge it into master and change the Opendatacam Dockerfiles to use the pre-compiled darknet.

vsaw, 2021-01-15:
Worked out like a charm. See https://hub.docker.com/r/opendatacam/opendatacam-darknet-base/tags?page=1&ordering=last_updated for the new base images.

Currently only the development run finished. The "master" release will finish in a few minutes. Then I can continue the refactoring on opendatacam side.

tdurand, 2021-01-15:
Awesome, you are officialy the master of github actions 🙌

vsaw, 2021-01-15:
Yeah I'm a big believer in testing and automation. They are great tools to manage complexity, increase confidence in releases and keep up development speed. 😄

tdurand, 2021-01-15:
exactly what was much needed for this project 😂

vsaw, 2021-01-15:
I could successfully run https://hub.docker.com/layers/opendatacam/opendatacam/development-nano/images/sha256-a985abb2748668e09d261ec9e067e702c9f17536ea5c892f43b29c78b2937786?context=explore. Therefore I will consider this as closed and done 👍

PR #366 (2021-01-15): refactor(Docker): Uses pre-built Darknet
Refactoring to speed up our Docker image build time.

Closes #358
closed 2021-01-15 by vsaw