opendatacam/opendatacam #363: Tesla M2090

zmalsberg, 2021-01-07:
Looking to acquire a Tesla M2090 eGPU for purposes of quickly analyzing recorded video data. Can anyone confirm if this card is compatible with the opendatacam stack? It supports CUDA compute 2.0.

I previously attempted to install on a Ubuntu 20.04 system with an old Quadro 6000 to no avail, I believe its CUDA version wasn't compatible, two attempts met with the same errors regarding missing components.

tdurand, 2021-01-07:
Hi, I think you should have a look at the list of the CUDA supported GPU on the nvidia website: https://developer.nvidia.com/cuda-gpus

Let us know your findings

tdurand, 2021-01-07:
PS: and yes it is Darknet , which is the piece of software that runs the neural network (YOLO) which requires a CUDA compatible GPU

zmalsberg, 2021-01-07:
Thank you Thibault for your speedy reply,

I see the M2090 on the legacy support list as v2.0 compatible. 
My only remaining question being is the corresponding cuDNN version high enough, or known to run the stack well?

Thanks,
Zach

tdurand, 2021-01-07:
I think you need:

CUDA >= 10.0
cuDNN >= 7.0

Taken from the darknet readme: https://github.com/alexeyab/darknet#requirements

closed 2021-05-22 by tdurand