opendatacam/opendatacam #364: Build fails due CUDNN (Xavier NX, JP 4.4.1) - ODC Darknet fork getting old?

Denizzje, 2021-01-07:
Greetings,

I have made a fresh install of Jetpack, and I notice the Darknet fork provided by Opendatacam fails to build in this configuration. Make file settings:

```
GPU=1
CUDNN=1
CUDNN_HALF=0
OPENCV=1
AVX=0
OPENMP=0
LIBSO=0
ZED_CAMERA=1 # ZED SDK 3.0 and above
ZED_CAMERA_v2_8=0 # ZED SDK 2.X
```

This is the error message:

```
./src/convolutional_layer.c: In function ‘cudnn_convolutional_setup’:
./src/convolutional_layer.c:286:24: error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTEST’ undeclared (first use in this function); did you mean ‘CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3’?
     int forward_algo = CUDNN_CONVOLUTION_FWD_PREFER_FASTEST;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3
compilation terminated due to -Wfatal-errors.
Makefile:162: recipe for target 'obj/convolutional_layer.o' failed
make: *** [obj/convolutional_layer.o] Error 1
```

I have also tried with CUDNN_HALF on 0, and I get the exact same build error. 

When I build darknet directly from the git, it builds just fine with the exact same configuration. It seems the ODC fork is getting a bit older now, it might be something in the latest jetpack which does not work well with that. Would it be possible to update the ODC darknet or will that break things?

tdurand, 2021-01-07:
Hi @Denizzje 

Yes, sorry, the latest jetpack doesn't play well with our darknet fork version, but good news is that the "latest" darknet should work out of the box: https://github.com/alexeyab/darknet

The diff of our fork is just a few comments to clean up the logs: https://github.com/opendatacam/darknet/pull/7/files

I'm aiming to spend time later this months to update OpenDataCam docker images + the fork in order to be compatible with latest jetpack.. In the meantime if you try it would be nice if you can let me know if it works

Thanks

Denizzje, 2021-01-08:
Hello @tdurand , 

I can confirm that the dev branch runs fine with the darknet straight from Alexay's git. Indeed some weird things in the console but it seems of no concequence ODC' s csvs and everything look fine.

20 fps on full Yolov4 in ODC' s simulation mode by the way. I had about the same FPS on 4.4 but the version I was running did not have simulation mode so it might change when I put my own model and videostream in it.

tdurand, 2021-01-08:
nice, if you put "simulation" in the config it is not running darknet.. but maybe you meant the demo.mp4 video.

the speed should be more of less the same, there is improvements possible though in the future by using something faster than darknet for inference:

- deepstream
- tkdnn : https://github.com/ceccocats/tkDNN

Denizzje, 2021-01-08:
Haha no, I am very smart and I actually did run the simulation in config. I thought it would actually run a demo video indeed Yolo on. It looked very convincing ;). Those were also the "weird things in the console" actually, that was loading in the array of detections from the json.

So I did not manage to get Yolo to work with the dev version but that is probably because of the different darknet input config settings which I must have done not correctly.

When running the release version, point it to the darknet straight from alexays git, it however works like normal without a hitch. I dont see anything different in the console.

So now I can confirm for real that it works ;).

Now with Yolo on for real, I get a glorious 5 fps with full fat Yolov4, and 30-40 with Yolov4-tiny.

tdurand, 2021-01-08:
okay! thanks for feedback, yes I'm gonna sort that out and update everything to latest darknet / yolo version at the end of the month 😉 .. great maybe if the console is now clean we will be able to get rid of the fork of darknet for good, and point to a particular commit of the main repo (as there is still no versioning ;-) )

vsaw, 2021-01-08:
Or we could repurpose the fork to just create a clean docker base for ODC? See https://github.com/opendatacam/opendatacam/issues/358

tdurand, 2021-01-08:
not sure if I got it , you mean having a github action on the darknet fork that will build a base docker image we then use as a base for the opendatacam docker auto build ? 😁

vsaw, 2021-01-09:
Yes. Building Darknet takes 5-7 minutes which we could shave off the docker build. So this repo becomes more a versioned package builder then a "source store".

icon244, 2021-03-03:
Hi,

I am trying to install the development branch on the jetson xavier with jetpack 4.4 and I am getting the following error.

Pheew ....! Something unexpected happened. This should be handled more gracefully. I am sorry. The culprit is:  Error: spawn 
/home/icon/Projects/opendatacam-development/opendatacam/darknet EACCES
    at Process.ChildProcess._handle.onexit (internal/child_process.js:267:19)
    at onErrorNT (internal/child_process.js:469:16)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'EACCES',
  code: 'EACCES',
  syscall: 'spawn /home/icon/Projects/opendatacam-development/opendatacam/darknet',
  path: '/home/icon/Projects/opendatacam-development/opendatacam/darknet',
  spawnargs: [
    'detector',
    'demo',
    'cfg/coco.data',
    'cfg/yolov4-416x416.cfg',
    'yolov4.weights',
    '/home/icon/Projects/opendatacam-development/opendatacam/opendatacam_videos/demo.mp4',
    '-ext_output',
    '-dont_show',
    '-dontdraw_bbox',
    '-json_port',
    8070,
    '-mjpeg_port',
    8090
  ]
}
Process YOLO started
(node:8909) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
First request on console stream

I pulled and compiled the latest release from alexeyab, which is working fine just on files. Are there any further changes I have to make?

Regards,

tdurand, 2021-03-03:
sorry didn't update OpenDataCam for Jetpack 4.4+ compat yet..

make sur you have this file ``yolov4-416x416.cfg` in the cfg/ folder: https://github.com/opendatacam/darknet/pull/7/files#diff-5a17ff24f270a5ad38d0da067ebf07b075c49e1bc85e0ccbf951ec1122d5ea80

We created it as the default is 600x600 for yolov4 full weight cfg and it was very slow ...

xunfangce, 2021-03-25:
> sorry didn't update OpenDataCam for Jetpack 4.4+ compat yet..
>
> make sur you have this file ``yolov4-416x416.cfg` in the cfg/ folder: [...]

hey i am really new to ODC, and i tried to deploy the app on my Jetson Nano today, finally it worked despite the difficulties of downloading the repo due to Chinese cyber-firewall. Anyhow, with all due respect, the instinct i have for now is , have you guys ever tried to accelerate the model with Nvidia's TensorRT. Compile with .engine files, it shouda work 5 or more times faster. 
Really impressed by your guys' work, looking forward to seeing new version of TRT supported application.

tdurand, 2021-03-25:
Thanks for your comment

> Nvidia's TensorRT. Compile with .engine files, it shouda work 5 or more times faster.

Darknet doesn't support TensorRT , what is why we don't use it yet.. And the speed from what I read isn't 5 times faster, but more like 30% faster.

There is an open issue to use deepstream instead of darknet to leverage this: https://github.com/opendatacam/opendatacam/issues/112

vsaw, 2021-12-08:
I will close this as there is now JP 4.6 released and therefore issues specific to JP 4.4.1 are considered outdated and obsolete.

closed 2021-12-08 by vsaw