opendatacam/opendatacam #365: OpenDataCam on Android : Call for beta tester

tdurand, 2021-01-10:
Hi all 🙌

Some great news to share with you for the start of this 2021, we got a grant a few months ago to start working on a OpenDataCam version running on Android.

This does not mean that the IoT version is deprecated... it just means that we are trying to expand the list of hardware you can run and play with OpenDataCam. As part of the funding we got we are also gonna update the OpenDataCam for Jetsons later this month.

We are ready to start sharing a broader beta test to detect and fix more bugs.. The goal of the beta test will also be to define the minimum CPU requirements in order to run the object detection at a decent speed.

We set-up a mailing list for the beta test, please register below and this week we will be in touch with the link to download the app beta on the google play along with a feedback form.

### [Link to Register for the beta test](http://eepurl.com/hlFnvv)

Feel free to ask anything, some more detailed info:

- The app code will be Open Source under MIT license after the release
- It currently runs Tiny YOLOv4 using [NCNN inference framework](https://github.com/Tencent/ncnn) from Tencent, on CPU .. but can run on mobile GPU too.. just right now NCNN is faster on CPUs

_OpenDataCam beta (android version)_
[image]

Thanks for the help 🙏

Thibault

laem, 2021-01-11:
Super news ! My S8 with a broken screen has been waiting for this in the dust :)

tdurand, 2021-01-12:
I just added a few emails to the beta and sent some instruction on how to get the app, but I noticed lots of email weren't valid google play accounts so if you can't download the app please reach out at opendatacam.project@gmail.com

jogiji, 2021-02-02:
@tdurand  Can you point to the source of this particular development.. I am interested in contributing to it.
Is it the mobile branch or the source is not committed yet ?

tdurand, 2021-02-02:
Hi , thanks for the message. It is not published yet.. working on cleaning things up but should go public in the next weeks.. in the meantime I add you to the beta

jogiji, 2021-02-27:
@tdurand any updates on source branch being published ?

tdurand, 2021-03-01:
working on it, should be done this week 😁

tdurand, 2021-03-04:
just published the OpenDataCam mobile repo & app on google play, thanks for all testers.. Would be nice if you can put some good rating ⭐⭐⭐⭐⭐ on the google play 😉 .

https://github.com/opendatacam/opendatacam-mobile

next step is to update the IoT version and the website.

closed 2021-03-04 by tdurand