opendatacam/opendatacam #367: No GSettings schemas are installed on the system

ItelBz, 2021-01-26:
Hi to all,
I've installed Opendatacam con a Nvidia Jetson, if I launch opendatacam with the demo mp4 file it works correctly but if I try yo use an rtsp stream it doesn't work, it returns the following error:

GLib-GIO-ERROR **: 10:58:03.148: No GSettings schemas are installed on the system

Opendatacam is installed with docker, I tried to install it witouth docker but I got the following error: 
error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTES, but this is not an issue for this problem.

How can I set the correct schema? I've tried also different rtsp streams.

Best regards

tdurand, 2021-01-26:
Hi, 

Yes there is issues with latest jetpack without docker: https://github.com/opendatacam/opendatacam/issues/364 , working on a new release of OpenDataCam coming soon.

With docker it should work.. never had this issue but I didn't work much with RTSP stream... What Gstreamer pipeline are your using ?

ItelBz, 2021-01-26:
Hi,
i'm using this string:

 "remote_cam": "rtsp://admin:password@192.168.1.205/Streaming/Channels/101"

I've used this string with the version 2.1.0 of opendatacam and it worked.

Thank you

tdurand, 2021-01-26:
Hum, maybe in https://github.com/opendatacam/opendatacam/issues/324 you'll find some answers 

Pinging @vsaw also on this  

I think the solution might be to read the rtsp stream using Gstreamer instead of relying on OpenCV to decode it.. but it is weird it was working with 2.1.0 and not working now.

vsaw, 2021-01-28:
Sorry, I've never encountered that error. But I've also never tried RTSP.

geologic, 2021-01-28:
@ItelBz: witch camera are you using? Maybe you are missing the port number in your rtsp URL

try "remote_cam": "rtsp://admin:password@192.168.1.205:554/Streaming/Channels/101"

ItelBz, 2021-02-03:
Hi,
I'm using an hikvision camera, i've solved using the following parameters:

`! latency=200 ! decodebin ! nvvidconv ! appsink`.

So the full string is:

`"remote_cam": "rtsp://admin:password@192.168.1.205:554/Streaming/Channels/101 ! latency=200 ! decodebin ! nvvidconv ! appsink"`

Now I've tried an Axis camera and got the same error, the pipeline working for Hikvision is not working for axis, I'll try different stream option.

closed 2021-02-16 by ItelBz

atennantmorpheus, 2021-07-01:
Just curious, has there been any progress on this or has anyone successfully fixed the issue after ths fact?   The streaming IP camera I'm using worked fine previously, but no longer works, thowing the GSettings schemas error. 

Thanks!