opendatacam/opendatacam #369: Tim Opendatacam Docker Image

vsaw, 2021-01-29:
The Opendatacam Docker Image seems to be quite large as the `.dockerignore` file seems to not filter out unnecessary things like `.git` or `spec` folders.

We should consider adding things like

```
# Git
.git
.gitignore

# Docker
Dockerfile
docker-compose.yml

# NPM dependencies
node_modules
```

vsaw, 2021-02-04:
See here for more docker best practices to implement at some point

- https://snyk.io/blog/10-best-practices-to-containerize-nodejs-web-applications-with-docker/
- https://dev.to/nodepractices/docker-best-practices-with-node-js-4ln4

vsaw, 2021-02-05:
Closed in #370

tdurand, 2021-02-06:
great work !

I think you leaved your weight file name instead of yolov4 : https://github.com/opendatacam/opendatacam/pull/370/files#diff-58aa082fcc0c51fa7c6e97622a009c2cf69f5e04ee034c8f455b15037698c48cR31

vsaw, 2021-02-06:
@tdurand good catch!

Please review the fix https://github.com/opendatacam/opendatacam/pull/371

tdurand, 2021-02-06:
no problem !

commit 5015730b (2021-02-05): Merge pull request #370 from vsaw/369-trim-docker-image

fix(docker): Trim Docker Image

Closes #369
PR #371 (2021-02-06): fix(docker): Fixes default neural network to right settings
Closes #369
closed 2021-02-06 by tdurand