opendatacam/opendatacam #372: Display ID on video

dylanstouls-bib, 2021-02-08:
Hi,

I would like to display the ID generated by ODC on a video in order to analyse which elements are causing a switch ID.
The desired result should be like the following :
[image]

**What I tried :**
I took a look at this old post :
https://github.com/opendatacam/opendatacam/issues/3#issuecomment-331155358

I did try to go to  http://localhost:8080/x-playground/tracker-test/index.html
But I got a 404 error

**Questions :** 
Is this feature still avaible ? 
Do I need to build an old version of ODC ?

vsaw, 2021-02-08:
That feature is not readily available. The closest we have is the path tracker which will assign the object a new color if the ID changes.

That being said, the ID *should* be forwarded to the Opendatacam front end. I would start investigating https://github.com/opendatacam/opendatacam/blob/master/components/canvas/engines/LiveViewEngine.js to see if the ID is in fact part of the `objectTracked` variable and then just change the displayed string .

DylanStouls, 2021-03-10:
Hi,
I'm replying with another account. Sorry the late reply.

Thank you @vsaw for putting me on the right way.
This was easy to do by changing[ these lines](https://github.com/opendatacam/opendatacam/blob/master/components/canvas/engines/LiveViewEngine.js#L73) : 

```
      // name -- text
      context.fillStyle = evaluateCSSVariable(colors.inverse);
      context.fillText(
        `${objectTrackedScaled.name}`,
```

 with 
```
      // name -- text
      context.fillStyle = evaluateCSSVariable(colors.inverse);
      context.fillText(
         ${objectTrackedScaled.id},
```

tdurand, 2021-05-22:
yeah.. add a visual debugging mode of ids would be nice.. but not planning to do it for v4

tdurand, 2021-05-22:
this is very handy when you work on improving the tracker

closed 2021-05-22 by tdurand