opendatacam/opendatacam #373: Not working from external IP

bobmorane83, 2021-02-15:
Hi,

Following #126 and #128.
I'am trying to access OpenDataCam from external IP (4G or even LAN through a router)

Here is my setup : 
Adlink Neon camera - USB <- USB/Ethernet Dongle -> Ethernet - Raspberry Pi with OpenWrt (router) - USB <- USB/Ethernet Dongle -> Macbook

The OpenWrt is configurated with firewall rules : 
Any IP:8080 from LAN or WAN -> Adlink camera IP:8080 (on LAN of course)
 
[image]

Addressing other services, on other IP:Port (with other firewall rules) are Ok.
Addressing standalone local server "$ serve -l 8080" is Ok.

Here is the problem : 
Addressing < Camera IP >:8080, the request never ends. No output, nothing.

But : 
Addressing < Camera IP >:8080/status is OK (JSON received)
Addressing < Camera IP >:8080/console is OK (same output as on console)

I've test the patch provided in #128, no change.

Any help ?
I can help for tests if nedded.

Thanks

bobmorane83, 2021-02-15:
Sorry,

Applying patch #128 gives me : 
[image]

But can't go further.

vsaw, 2021-02-15:
Can you copy the complete log output here? From this snippet I'm not sure if darknet is running. If darknet can not be started you will be stuck on the init screen forever.

bobmorane83, 2021-02-15:
Yes, Darknet is running ok.
Acceding to opendatacam without the router (directly on same LAN) is Ok, and running as expected.
Adding the router cause the problem.

Here is the full log.

```
adlink@adlink-desktop:~/XXXXXXXX/opendatacam$ npm run start

> OpenDataCam@3.0.1 start /home/adlink/XXXXXXXX/opendatacam
> PORT=8080 NODE_ENV=production node server.js

-----------------------------------
-     Opendatacam initialized     -
- Config loaded:                  -
{
  "OPENDATACAM_VERSION": "3.0.1",
  "PATH_TO_YOLO_DARKNET": "/home/adlink/XXXXXXXX/darknet",
  "VIDEO_INPUT": "usbcam",
  "NEURAL_NETWORK": "yolov4-tiny",
  "VIDEO_INPUTS_PARAMS": {
    "file": "opendatacam_videos/demo.mp4",
    "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1 ! videoconvert ! appsink",
    "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
    "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
    "local_cam": "pylonsrc imageformat=rgb8 ! videoconvert ! appsink",
    "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink"
  },
  "TRACKER_SETTINGS": {
    "objectMaxAreaInPercentageOfFrame": 80,
    "confidence_threshold": 0.2,
    "iouLimit": 0.05,
    "unMatchedFrameTolerance": 5
  },
  "COUNTER_SETTINGS": {
    "minAngleWithCountingLineThreshold": 5,
    "computeTrajectoryBasedOnNbOfPastFrame": 5
  },
  "VALID_CLASSES": [
    "*"
  ],
  "DISPLAY_CLASSES": [
    {
      "class": "bicycle",
      "hexcode": "1F6B2"
    },
    {
      "class": "person",
      "hexcode": "1F6B6"
    },
    {
      "class": "truck",
      "hexcode": "1F69B"
    },
    {
      "class": "motorbike",
      "hexcode": "1F6F5"
    },
    {
      "class": "car",
      "hexcode": "1F697"
    },
    {
      "class": "bus",
      "hexcode": "1F68C"
    }
  ],
  "PATHFINDER_COLORS": [
    "#1f77b4",
    "#ff7f0e",
[... 98 lines cut ...]
  37 yolo
[yolo] params: iou loss: ciou (4), iou_norm: 0.07, obj_norm: 1.00, cls_norm: 1.00, delta_norm: 1.00, scale_x_y: 1.05
Total BFLOPS 6.910 
avg_outputs = 310203 
 Allocate additional workspace_size = 26.22 MB 
Loading weights from yolov4-tiny.weights...Demo
net.optimized_memory = 0 
mini_batch = 1, batch = 1, time_steps = 1, train = 0 
nms_kind: greedynms (1), beta = 0.600000 
nms_kind: greedynms (1), beta = 0.600000 
Done! Loaded 38 layers from weights-file 

 seen 64, trained: 0 K-images (0 Kilo-batches_64) 
video file: v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1 ! videoconvert ! appsink
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
setvideoresolution
Video stream: 2592 x 1944 
Objects:

 JSON-stream sent. 

FPS:0.0          AVG_FPS:0.0
Objects:

```

vsaw, 2021-02-16:
Thank you for the logs. I have a theory why Opendatacam is not handling your NAT setup correctly. I believe what happens is the following:

1. Opendatacam listens on any IP address/hostname for incoming connections on Port 8080.
2. The first client connecting to `<odc>:8080/` or `<odc>:8080/start` starts the Darknet process (see [server.js](https://github.com/opendatacam/opendatacam/blob/development/server.js#L145))
3. Opendatacam then tries to connect to Darknet using the `<odc>` IP address/hostname from your browser (see [Opendatacam.js](https://github.com/opendatacam/opendatacam/blob/development/server/Opendatacam.js#L792)). This works fine **unless** you use NAT.
4. Because now Opendatacam tries to connect to Darknet on your public IP/hostname ports 8070 and 8090. However as there are no port forwarding rules setup, the connection will fail.

Therefore, to fix this you have two options:

1. Setup port forwarding for 8070 and 8090
2. Fix the Darknet hostname in https://github.com/opendatacam/opendatacam/blob/development/server/Opendatacam.js#L792 to `localhost` (I have not tried this, but something like this should work)

bobmorane83, 2021-02-16:
Thanks for your answer.

Option 1 :
Here is the port forwarding rules : 
[image]

Addressing < IP >:8070 : Ok, I can see JSON output from Darknet
Addressing < IP >:8090 : Ok, I can see the video stream

But, the opendatacam UI still does not show up :( 

Option 2 (Option 1 is still on):
 
The opendatacam UI show up and working correctly, but can't see the video stream :(

I'am sure we're close :)
May be we need to do the same fix for the video stream ?

Thanks,

bobmorane83, 2021-02-16:
I modified server.js with : 

 express.get('/webcam/stream', (req, res) => {
    const urlData = getURLData(req);
    // Proxy MJPEG stream from darknet to avoid freezing issues
    return new MjpegProxy(`http://localhost:8090`).proxyRequest(req, res);
  });

And it works.
 
But I'am sure there is a better way to do that !!!

vsaw, 2021-02-16:
Sorry I forgot about the MJPEG Proxy! For now I believe we could fix it to `localhost`, like you did, to get Opendatacam working behind NAT.

Long term we should consider making this configurable for more elaborate setups where Darknet is running on a different machine.

vsaw, 2021-02-16:
@bobmorane83 I've created a follow up issue to fix this. If your problem is now solved feel free to close this here.

bobmorane83, 2021-02-16:
Thanks for your support.

closed 2021-02-16 by bobmorane83