opendatacam/opendatacam #374: Make Opendatacam reachable behind NAT

vsaw, 2021-02-16:
~~Thank you for the logs. I have a theory why~~ Opendatacam is not handling your NAT setup correctly. ~~I believe what~~ happens is the following:

1. Opendatacam listens on any IP address/hostname for incoming connections on Port 8080.
2. The first client connecting to `<odc>:8080/` or `<odc>:8080/start` starts the Darknet process (see [server.js](https://github.com/opendatacam/opendatacam/blob/development/server.js#L145))
3. Opendatacam then tries to connect to Darknet using the `<odc>` IP address/hostname from your browser (see [Opendatacam.js](https://github.com/opendatacam/opendatacam/blob/development/server/Opendatacam.js#L792)). This works fine **unless** you use NAT.
4. Because now Opendatacam tries to connect to Darknet on your public IP/hostname ports 8070 and 8090. However as there are no port forwarding rules setup, the connection will fail.

Therefore, to fix this ~~you~~ we have two options:

1. Setup port forwarding for 8070 and 8090 *(not preferred)*
2. Fix the Darknet hostname in https://github.com/opendatacam/opendatacam/blob/development/server/Opendatacam.js#L792 *and https://github.com/opendatacam/opendatacam/blob/development/server.js#L188* to `localhost`

_Originally posted by @vsaw in https://github.com/opendatacam/opendatacam/issues/373#issuecomment-779727816_

zeljkomarinkovic, 2021-04-16:
Can confirm that latest fix works, though have another issue where data is not stored also if you try to download counter JSON data id is always null

[image]

vsaw, 2021-04-16:
Is it just the JSON?

zeljkomarinkovic, 2021-04-16:
same for all of them

vsaw, 2021-04-16:
I have an idea but will need your help to confirm it.

## Theory
Before https://github.com/opendatacam/opendatacam/pull/347 we used the MongoDB internally created `_id` field. With the merge in place, we create the recording ID in ODC and store it in the newly created `id` field.

However we do not run any migrations on MongoDB. Therefore, existing recordings will not have the `id` field and may cause your error.

## Verification of Theory

- Did you recently rebase/merge the development branch?
- Are you working on old recordings?
- If you create a new recording. Is the link still broken?

zeljkomarinkovic, 2021-04-17:
> Did you recently rebase/merge the development branch?
No, I'm running ODC on a master branch
> Are you working on old recordings?
No, straight from camera into device
> If you create a new recording. Is the link still broken?
Yes, links are still broken

zeljkomarinkovic, 2021-04-17:
> ## Verification of Theory
> * Did you recently rebase/merge the development branch?

No, running ODC on a master branch

> * Are you working on old recordings?

No, device is processing data from ip camera

> * If you create a new recording. Is the link still broken?

Yes, links are still broken, when I create new recording

vsaw, 2021-04-20:
Allright thanks! Then I'll have to do some digging to figure out what happens here.

vsaw, 2021-04-20:
@zeljkomarinkovic I've got another theory. I blieve this time it has nothing to do with the NAT.

Because I noticed a discrepancy how the recordingID is inserted into the React Component. Compare the following to lines:

- Running Recording: https://github.com/opendatacam/opendatacam/blob/master/components/main/DataView.js#L39
- Finished Recording: https://github.com/opendatacam/opendatacam/blob/master/components/main/DataView.js#L51

Can you please check if the link is working on a finished recording?

zeljkomarinkovic, 2021-04-20:
It was my fault, tried to install curl on that machine and somehow screwed up mongod.services (i think libcurl is culprit there). Reinstalled mongodb and works just fine. 
Sorry for all trouble i caused and not checking obvious first

vsaw, 2022-02-14:
Closed via https://github.com/opendatacam/opendatacam/pull/527

closed 2022-02-14 by vsaw