opendatacam/opendatacam #376: screenshot of frame when one car crosses the specified line

Iman1221, 2021-03-03:
Thank you for the project provided.
How can I get an screenshot of frame when one car crosses the specified line?
Is there any way?

vsaw, 2021-03-03:
There was a discussion about this where this in https://github.com/opendatacam/opendatacam/issues/170.

Long story short, it was decided by the project owners that this is against Opendatacams "privacy by design" principle. However, there's nothing stopping you from forking Opendatacam and implementing it yourself. ;-)

haviduck, 2021-03-14:
since odc calls darknet directly, you gonna have to work with the api and grab your frame from the mjpeg stream on the relevant sse event.

vsaw, 2021-03-24:
@haviduck Thanks! That's exactly what I do, works like a charm :-)

closed 2021-05-22 by tdurand